roboticslibrary/rl #8: Closed chain kinematic robot model

falfab, 2019-05-28:
Does RL support the creation of robot models with closed kinematic chains?

If not, is it possible to implement the robot model as multiple open kinematic chains, and then easily extend or substitute the kinematic solver in order to satisfy the closed kinematic constraints?

rickertm, 2019-06-02:
RL currently only supports kinematic trees without closed loops. Depending on what features you need (forward position, inverse dynamics, ...), the chapter 8 "Closed Loop Systems" in the book "Rigid-Body Dynamics Algorithms" gives an outlook on how this could be added.

rl::mdl::Model currently includes gammaPosition and gammaVelocity matrices for handling kinematics with a static connection between position variables q and independent position position variables y. If your model has an analytical solution (Fig. 8.3 on page 163), it should be possible to integrate this here with a suitable gamma function.

falfab, 2019-06-04:
Thank you for the suggestion, I will study it and let you know. For the moment I close the issue/question.

closed 2019-06-04 by falfab