roboticslibrary/rl #9: rlSplineTranslationTest Fail

falfab, 2019-05-28:
LSB:
```
LSB Version:	core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.5 LTS
Release:	16.04
Codename:	xenial
```

Kernel:
```
Linux Jessy 4.15.0-50-generic #54~16.04.1-Ubuntu SMP Wed May 8 15:55:19 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

GCC version: 
```
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

ctest output:
``
...
18: Test command: /home/mesa/rl/rl/build/bin/rlSplineTranslationTest
18: Test timeout computed to be: 9.99988e+06
18: 1.7*x^5 -2.5*x^2 1.5*x^1 
18: 1.7*x^5 10.2*x^4 24.48*x^3 26.876*x^2 13.1256*x^1 2.43014*x^0 
18: rlSplineTranslationTest: /home/mesa/rl/rl/src/rl/math/Polynomial.h:351: T rl::math::Polynomial<T>::operator()(const Real&, const size_t&) const [with T = Eigen::Array<double, -1, 1>; rl::math::Real = double; std::size_t = long unsigned int]: Assertion `x < this->upper() + FUNCTION_BOUNDARY' failed.
18/40 Test #18: rlSplineTranslationTest ......................***Exception: Other  0.08 sec
...
```

rickertm, 2019-06-02:
Thank you for reporting this. Should be fixed by ef307431bc5a83bfdf2dba6a98fd4f2847ad20a1.

falfab, 2019-06-03:
@rickertm I can confirm the modifications done in https://github.com/roboticslibrary/rl/commit/ef307431bc5a83bfdf2dba6a98fd4f2847ad20a1 solve the issue. But, in order to test it, I had to manually add the modifications of https://github.com/roboticslibrary/rl/commit/ef307431bc5a83bfdf2dba6a98fd4f2847ad20a1 to the sources of 0.7 Release, since the master branch does not compile on mine machine.

Since this issue is solved I'm going to close it. I will open a new one describing my compilation problem.

closed 2019-06-03 by falfab