roboticslibrary/rl #10: Kinematics forward acceleration wrong?

dfeather, 2019-08-05:
Hello, I'm trying to use the awesome library for some project.
And I found that the algorithm in **forwardAcceleration()** in **mdl::Kinemtics** seems not working correctly.
I did not look into the algorithm, but I found this computing uses the world gravity param in model xml.
```
		<world id="world">
			<rotation>
				<x>0</x>
				<y>0</y>
				<z>0</z>
			</rotation>
			<translation>
				<x>0</x>
				<y>0</y>
				<z>0</z>
			</translation>
			<g>
				<x>0</x>
				<y>0</y>
				<z>12.345</z>
			</g>
		</world>

```
When I changed the **g** element in the model xml, the computing result will change.
This is strange, because I Believe kinematics computing has nothing to do with gravity acceleration(dynamics computing use it).

I also try to verify the computing result with the [KDL](https://github.com/orocos/orocos_kinematics_dynamics) library, and it doesn't match...

Am I correct?
I'm using rl-0.7.0

rickertm, 2019-08-06:
The forwardAcceleration() function in rl::mdl::Kinematic updates the accelerations in the model's elements in a forward iteration, where gravity is modeled and stored as acceleration in the World element. For a kinematic computation, the gravity currently has to be set to zero before calling the function, whereas for a dynamic computation it requires an appropriate gravity acceleration.

In order to make this less error-prone, I have updated this behavior in f7039941a44ba8c39e9e6cc7a63c1862d88ba770. Gravity is now just a member variable of the World element and the element's acceleration is initialized depending on the function that is called, i.e., zero for forwardAcceleration() and equal to gravity for inverseDynamics() and forwardDynamics(). Let me know if this helps.

dfeather, 2019-08-07:
Still something I don't understand.
I set the gravity to zero.
And the computing result mismatches the one in KDL library.

What I have done is that I create the same serial kinematic chain in both RL and KDL and try to check some computing result.
The reuslt of forwardPosition, forwardVelocity, Jacobian computing matches, but the derivative of jacobian and forwardAcceleration mismatch.

I can give you a clue.
Suppose we have a link on a single revolute joint(one dof simple pendulum）
the only joint rotates at constant angular velocity.
for RL, the computed operational acceleration is zero.
for KDL, the computed operational aceeleration contains a centripetal acceleration which relies on the joint velocity and position.

Waiting for your reply...thanks...

rickertm, 2019-08-07:
Is this with the current master branch or 0.7.0? Could you provide a full example demonstrating this, including code, model file, and output?

dfeather, 2019-08-09:
I'm using 0.7.0

There are two things.

- Before calling **calculateJacobian**, I must call **forwardPosition**, otherwise the result is not correct(seems always the jacobian at zero position). Is this Intentional?

- The result of jacobian direvative and operational acc mismatch.

Underneath is the example.

model file: **one_joint_kin.xml**
```
<?xml version="1.0" encoding="UTF-8"?>
<rlmdl xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="rlmdl.xsd">
	<model>
		<manufacturer>for_test</manufacturer>
		<name>for_test</name>
		<world id="world">
			<rotation>
				<x>0</x>
				<y>0</y>
				<z>0</z>
			</rotation>
			<translation>
				<x>0</x>
				<y>0</y>
				<z>0</z>
			</translation>
			<g>
				<x>0</x>
				<y>0</y>
				<z>0.0</z>
			</g>
		</world>
		<body id="body0">
		</body>
		<frame id="frame0"/>
		<frame id="frame1"/>
		<body id="body1">
			<cm>
				<x>-1.0</x>
				<y>0</y>
				<z>0</z>
			</cm>
			<i>
				<xx>1</xx>
				<xy>0</xy>
				<xz>0</xz>
				<yy>1</yy>
				<yz>0</yz>
				<zz>1</zz>
			</i>
			<m>2</m>
		</body>
		<frame id="frame2"/>
		<fixed id="fixed0">
			<frame>
				<a idref="world"/>
				<b idref="body0"/>
			</frame>
			<rotation>
				<x>0</x>
				<y>0</y>
				<z>0</z>
			</rotation>
			<translation>
				<x>0</x>
				<y>0</y>
				<z>0</z>
			</translation>
		</fixed>
		<fixed id="fixed1">
			<frame>
				<a idref="body0"/>
				<b idref="frame0"/>
			</frame>
			<rotation>
				<x>0</x>
				<y>0</y>
				<z>0</z>
			</rotation>
			<translation>
				<x>0</x>
				<y>0</y>
				<z>0</z>
			</translation>
		</fixed>
		<revolute id="joint0">
			<frame>
				<a idref="frame0"/>
				<b idref="frame1"/>
			</frame>
			<axis>
				<x>0</x>
				<y>0</y>
				<z>1</z>
			</axis>
			<max>165</max>
			<min>-165</min>
			<speed>140</speed>
		</revolute>
		<fixed id="fixed2">
			<frame>
				<a idref="frame1"/>
				<b idref="body1"/>
			</frame>
			<rotation>
				<x>0</x>
				<y>0</y>
				<z>0</z>
			</rotation>
			<translation>
				<x>1</x>
				<y>0</y>
				<z>0</z>
			</translation>
		</fixed>
		<home>
			<q unit="deg">0</q>
		</home>
	</model>
</rlmdl>
```

rl code:
```

#include <iostream>
#include <rl/math/Transform.h>
#include <rl/math/Unit.h>
#include <rl/mdl/Kinematic.h>
#include <rl/mdl/Model.h>
#include <rl/mdl/XmlFactory.h>

int main(int argc, char** argv)
{
	rl::mdl::XmlFactory factory;
	rl::mdl::Kinematic * kinematics = dynamic_cast<rl::mdl::Kinematic*>(factory.create("one_joint_kin.xml"));
	rl::math::Vector q(1),q_dot(1),q_dotdot(1);
	q << 2.0;
	q_dot << 3.0;
	//always set joint acc to zero.
	q_dotdot << 0.0;

	kinematics->setPosition(q);
	kinematics->setVelocity(q_dot);
	kinematics->setAcceleration(q_dotdot);
	kinematics->forwardAcceleration();

	auto operational_acc = kinematics->getOperationalAcceleration(0);
	std::cout<<"q "<<q<<"\n";
	std::cout<<"q_dot "<<q_dot<<"\n";
	std::cout<<"operational acc\n";
	std::cout<<operational_acc.matrix().transpose()<<"\n";

	kinematics->setPosition(q);
	// we must add this forwardPosition call before calclateJacobian,
	// or the jacbobian result seems always the jacobian at zero position.
        // you can try comment it.
	kinematics->forwardPosition();
	kinematics->calculateJacobian();
	auto jacobian = kinematics->getJacobian();
	std::cout<<"jacobian\n";
	std::cout<<jacobian<<"\n";

	kinematics->setPosition(q);
	kinematics->setVelocity(q_dot);
	kinematics->calculateJacobianDerivative();
	auto jacobian_deriv = kinematics->getJacobianDerivative();
	std::cout<<"jacobian_deriv\n";
	std::cout<<jacobian_deriv.transpose()<<"\n";

	return 0;
}
```

rl output:
```
q 2
q_dot 3
operational acc
0 0 0 0 0 0
jacobian
-0.909297
-0.416147
        0
        0
        0
        1
jacobian_deriv
0 0 0 0 0 0
```
KDL code:
```

#include <frames_io.hpp>
#include <chainfksolverpos_recursive.hpp>
#include <chainjnttojacsolver.hpp>
#include <chainfksolvervel_recursive.hpp>
#include <chainjnttojacdotsolver.hpp>

using namespace KDL;

int main()
{
	//KDL do not use model file. we can directly model the chain here.
	auto f_tip = KDL::Frame();
	f_tip.p.x(1.0);
	KDL::Segment segment1(KDL::Joint(KDL::Joint::RotZ),f_tip);

	KDL::Chain chain;

	chain.addSegment(segment1);
	///////////////////

	KDL::JntArray q(1),q_dot(1),q_dotdot(1);
	KDL::ChainJntToJacSolver jac_solver(chain);
	KDL::Jacobian jac(1);
	ChainJntToJacDotSolver jac_dot_solver(chain);
	KDL::JntArrayVel q_vel(1);
	KDL::Jacobian jac_dot(1);
	// KDL do not have a method to compute operational acc directly.
	// We just get jac and jac_dot
	// and compute the acc = jac_dot * q_dot + jac * q_dotdot

	//helper data
	q.data<<2.0;
	q_dot.data<<3.0;
	q_dotdot.data<<0.0;

	q_vel.q = q;
	q_vel.qdot = q_dot;

	jac_solver.JntToJac(q,jac);
	jac_dot_solver.JntToJacDot(q_vel,jac_dot);
	std::cout<<"q "<<q.data<<"\n";
	std::cout<<"q_dot "<<q_dot.data<<"\n";
	std::cout<<"operational acc\n";
	std::cout<<(jac_dot.data * q_dot.data + jac.data * q_dotdot.data).transpose()<<"\n";
	std::cout<<"jacbian\n";
	std::cout<<jac.data<<"\n";
	std::cout<<"jacbian_dot* q_dot\n";
	std::cout<<(jac_dot.data * q_dot.data).transpose()<<"\n";

}
```

KDL output:
```
q 2
q_dot 3
operational acc
 3.74532 -8.18368        0        0        0        0
jacbian
-0.909297
-0.416147
        0
        0
        0
        1
jacbian_dot* q_dot
 3.74532 -8.18368        0        0        0        0
```

rickertm, 2019-09-18:
Thank you for providing a full example and bringing this to my attention. d6884c761e3f8afd2586d3dbeb1cb8500c7987f4 should fix the transformation to world frame for the Jacobian derivative calculation:
```
jacobian_deriv
 3.74532 -8.18368        0        0        0        0
```

Calling `forwardPosition()` before `calculateJacobian()` or `calculateJacobianDerivative()` is only required, when the result should use the world frame as reference (`inWorldFrame = true`). For an end-effector reference frame, this is not required and therefore optional.

dfeather, 2019-10-14:
OK, thanks.
it seems OK. I will try this ASAP.