roboticslibrary/rl #12: Cannot compile some demos in both Debug and Release due to ode common

guibatun, 2019-11-29:
Hi,
I am building RL from source. using windows, x64, and the Visual Studio 2017 compiler (VC 14.1). I need to build both Debug and Release versions.
I built all dependencies hopefully properly, by using the links to all the proper dependency versions.
I am building tests and demos.  
It seems the common.h file in the ode include directory is throwing errors which affect some projects like rlPLanDemo which is the one I need.
Could you see what is the problem? I am not sure where the dDouble type has been defined actually.

[image]
[common.zip](https://github.com/roboticslibrary/rl/files/3905735/common.zip)

Thanks.
I actually don't need to deal with all this, I would love if you have a windows installer that can also install the debug version of the libraries?

rickertm, 2019-12-01:
When using ODE's CMake build files, it will default to double precision on 64-bit systems, wheres 32-bit systems use single precision. This can be overridden with the CMake option `ODE_DOUBLE_PRECISION`. The resulting library files will be named `ode_double.lib` or `ode_single.lib` respectively. The CMake config file included in the installation will add `dIDEDOUBLE` or `dIDESINGLE` to the `INTERFACE_COMPILE_DEFINITIONS`.

As most Linux versions still ship an older ODE version without CMake config file, RL will link against ODE during configuration to detect which variant the linked library provides via a call to `dCheckConfiguration("ODE_double_precision")`. It then adds either `dDOUBLE` or `dSINGLE` to `ODE_DEFINITIONS` and uses this for `target_compile_definitions(sg ...)`.

ODE's `common.h` verifies that only one of these definitions has been added to the compiler. It looks like in your case, both are defined? Could you please provide some more details on how you built ODE, e.g., command line history, build logs, verbose output of make?

Thanks for the suggestion, I'll look into including both release and debug builds in the next installer. This may however lead to a large increase in file size due to the debug symbols.

guibatun, 2019-12-02:
Hi, I downloaded the ode source files from https://bitbucket.org/odedevs/ode/downloads/ode-0.16.tar.gz as per the link in the cmakelists file in rl's 3rd party section.
I then commented out the URL and MD5 sections since I got the ode sources in a tar.gz file myself due to restrictions we cannot allow a program through the firewall. I then let the build magic run through all the 3rd Party libraries (all gathered manually thus) and ode seems to be the only problem. 
The CMAKE_SIZEOF_VOID_P EQUAL 4 line must have worked, since my libraries are called ode_double.lib and ode_doubled.lib but when it gets to the install phase, they somehow all end up installed in C Program Files x86 which I find weird and am unsure what causes this.
Attached is the CMakeLists and the tar.gz file I used for creating the project. It builds just fine!
I also attached my build folder after install...Are the cmake output log files in there somewhere?
I am unsure where I can find in the RL project which components adds dDouble and which add dSingle actually.

Thanks.

[ode_build.zip](https://github.com/roboticslibrary/rl/files/3911578/ode_build.zip)

[ode_src.zip](https://github.com/roboticslibrary/rl/files/3911548/ode_src.zip)

rickertm, 2019-12-02:
Thank you for providing the additional information and the build directory. I couldn't find any issues in the generated files and was able to compile RL's current master branch against the files included in the debug build (with a recreated directory structure).

`build\ode-prefix\src\ode-build\include\precision.h` is correctly generated from `precision.h.in` with `dDOUBLE` in the place of `@ODE_PRECISION@`.

`build\ode-prefix\src\ode-build\CMakeFiles\Export\lib\cmake\ode-0.16.0\ode-export.cmake` correctly includes `dIDEDOUBLE` in `INTERFACE_COMPILE_DEFINITIONS`.

Did you make sure, that `ode_doubled.dll` is included in your `PATH` when running CMake? If not, `check_c_source_runs` in `FindODE.cmake` will fail to execute and add `dSINGLE` to `ODE_DEFINITIONS`. You can add a debug output after line 142 to verify this:
```
message(STATUS "ODE_DEFINITIONS: ${ODE_DEFINITIONS}")
```
You can also check the file `CMakeFiles\CMakeError.log` in RL's build directory. There should not be an entry for a test with `dCheckConfiguration(...)` and a return value of `FAILED_TO_RUN`.

guibatun, 2019-12-03:
Hi Markus, thanks for the pointers. I had not set the dlls in the Path, but that was not the issue. There was still the same problem.
I cannot understand why but the check_c_source_runs line in FindODE.cmake can never compile it seems, since ODE_DOUBLE_PRECISION is never '1'.
If I understand correctly, it does not matter what the main function returns, we just check if the code compiles and links. It does not in my case. I tried this
 check_c_source_runs("
	#include <ode/ode.h>
	int main(){return 1; }
	
" ODE_DOUBLE_PRECISION) and still never was ODE_DOUBLE_PRECISION==1 ( and thus ODE_DEFINITIONS was set to DdSINGLE) . However, I forced the ODE_DEFINITIONS to be -DdDOUBLE:
if(ODE_DOUBLE_PRECISION)
	set(ODE_DEFINITIONS -DdDOUBLE)
else()
	set(ODE_DEFINITIONS -DdDOUBLE)
endif()
and then I was able to compile and run the rlPlanDemo without a problem.
I don't really understand where it was set as DdDOUBLE then DdSINGLE, as it seems in my case all the calls to FindODE.cmake led to set(ODE_DEFINITIONS -DdSINGLE).

rickertm, 2019-12-04:
I've updated the check for the configured ODE precision in f8a4eb8948a415658aeccdd906e2cc45d5515a33. For recent ODE versions, this should skip the call of `dCheckConfiguration()`. Let me know if this works for you as well.

guibatun, 2019-12-17:
Confirmed this fixed it for me and -dDouble is found.

closed 2019-12-17 by guibatun