roboticslibrary/rl #13: Building rlPlanDemo stand-alone

guibatun, 2020-01-10:
Hi  Markus,
I will post this issue here since it might benefit other users.
As mentionned to you earlier I am trying to build rlPlanDemo without building all of RL (which succeeded in my case. I can build rlplandemo only when it is part of the RL build). Thus I am trying to know which elements of the RL CMakeLists.txt I am missing, to make a compact build based on rlPlanDemo.
I attach here the cmakelists.txt I am using, with the recommendation you mentionned about  adding the lines 
set(CMAKE_AUTOMOC ON)
set(CMAKE_INCLUDE_CURRENT_DIR ON)
It has not worked for me and produced other strange problems rl:plan (see included build_errors.txt). I also attach the build artefacts.
[build.zip](https://github.com/roboticslibrary/rl/files/4044536/build.zip)
[rlPlanDemo.zip](https://github.com/roboticslibrary/rl/files/4044541/rlPlanDemo.zip)

guibatun, 2020-01-10:
Hi Markus.
I solved this issue. It seems I had cloned the master branch which made use of functionality not found in  the RL 0.7.0 release ;)
I managed to build the original rlplandemo with that Cmakelist file. However, when I try with my own version of the demo, which again works when it is within the RL project, I get all kinds of winsock and w2def errors (see attached).
Do you get what this is related to?
[build_errors.txt](https://github.com/roboticslibrary/rl/files/4046063/build_errors.txt)

rickertm, 2020-01-10:
Please try adding the following definitions to your `add_definitions()` command on line 5 of your CMakeLists.txt:
```
-D_WIN32_WINNT=0x501
-DNOMINMAX
-DWIN32_LEAN_AND_MEAN
```