roboticslibrary/rl #14: Using scene to load and SimpleScene to check collision

svenkatachalam, 2020-03-03:
I am trying to replicate the provided code snipets on loading a scene with rl::sg::so::scene and then performing dynamic_cast to convert the scene object into SimpleScene. 
Code: rl::sg::SimpleScene* simpleScene = dynamic_cast<rl::sg::SimpleScene*>(&scene) 
As both doesnt hold Base and derived class relationship between each other, this conversion fails (Getting null object). My idea is to find if two Robots collide with each other in the given scene environment.

rickertm, 2020-03-07:
In order to perform collision checking with a rl::sg::SimpleScene, please make sure to use an engine that is derived from this class. rl::sg::so::Scene (an Open Inventor/Coin3D abstraction) only supports 3D visualization and does not implement [rl::sg::SimpleScene.](http://doc.roboticslibrary.org/0.7.0/d7/d10/classrl_1_1sg_1_1_simple_scene.html).

Please use either the Bullet, FCL, ODE, PQP, or SOLID engine for this:

```cpp
rl::sg::solid::Scene scene;
rl::sg::XmlFactory factory;
factory.load("scene.xml", &scene);
bool isColliding = scene.isColliding();
bool areColliding = dynamic_cast<rl::sg::SimpleScene*>(&scene)->areColliding(
  scene.getModel(0), scene.getModel(1)
);
```

rickertm, 2020-03-09:
Since 35dc4a37de4f180da8b0422c8e64e25f2bc94c51 the `dynamic_cast` is no longer necessary and you can also use
```cpp
bool areColliding = scene.areColliding(scene.getModel(0), scene.getModel(1));
```