roboticslibrary/rl #16: Black Screen on MacOS Catalina

gi-diana, 2020-03-17:
I have installed RL via homebrew and when I run the first example
 `/usr/local/bin/rlViewDemo /usr/local/share/rl-0.7.0/examples/rlsg/unimation-puma560_boxes.xml
`
and the other examples too I get only a black windows 
when I try to resize and move the window sometimes I can see something behind the black screen 

[image]

How can I do to get things working ?

rickertm, 2020-03-17:
This is a known issue in Coin3D's SoQt library when using a HiDPI display such as Apple's Retina displays, see coin3d/soqt#1. In this case, the size of the OpenGL window does not match the window size.

I have tried to solve this in a patch for SoQt (see roboticslibrary/soqt@72c62073656780df44e63395596cf1f6966eb6ed), however this does not seem to fix all related issues, especially when moving a window from HiDPI to non-HiDPI displays. On a device with a single Retina display however, this patch should work.

Could you try to build the SoQt dependency with this patch on your machine and link RL against this to see if it solves your issue?

gi-diana, 2020-03-18:
I've built the SoQt dependency in the branch with your patch and then rebuilt RoboticsLibrary and now is working flawless, Thank You !
[image]

closed 2020-03-18 by gi-diana