roboticslibrary/rl #17: using URDF robot description

engr-zubair, 2020-04-24:
hi there, please let me know how I can use a URDF file description of robot for generating kinematic model , I have look down your rl::mdl:: urdffactory but the format is not clear how to create and load the urdf file instead of xml format , 
if you can please write down the command syntax. for using say robotmodel.urdf file stored at rl/share/mdl folder where other models xml files are stored. 
code syntax required for windows c++ visual studio 2017 
thanks for your help .! 

code syntax required for windows c++ visual studio 2017 
rl::mdl::UrdfFactory urdf_factory;
rl::mdl::Kinematic* kinematics = dynamic_cast<rl::mdl::Kinematic*>(urdf_factory.create("C:\\Program Files\\RoboticsLibrary\\0.7.0\\MSVC\\14.1\\x64\\share\\rl0.7.0\\examples\\rlmdl\\robot_model.urdf"));

rickertm, 2020-04-24:
The `Factory::create` function on the master branch now returns a `std::shared_ptr<Model>` instead of the `Model*` in the 0.7 branch. The `dynamic_cast` therefore has to be replaced with a `std::dynamic_pointer_cast`. You can also have a look at `rlInversePositionDemo.cpp` for an additional example.
```cpp
rl::mdl::UrdfFactory urdf_factory;
std::shared_ptr<rl::mdl::Kinematic> kinematics = std::dynamic_pointer_cast<rl::mdl::Kinematic>(
	urdf_factory.create("C:\\Program Files\\RoboticsLibrary\\0.7.0\\MSVC\\14.1\\x64\\share\\rl-0.7.0\\examples\\rlmdl\\robot_model.urdf")
);

engr-zubair, 2020-04-27:
hi , I have used the code as mentioned but it only works with the master branch and source installation of current RL lib. 
but I prefer to use rl-0.7 branch as it comes with pre-built bin for 3rd party libraries as I need to use other functionality like boost , eigen and qt run time etc ,it was installed in win bin rl-0.7 branch. 

what is the way to incorporate current master branch with Win-bin installation rl-0.7 ?
thanks.

rickertm, 2020-04-27:
For the rl-0.7 branch the code should look like this (with shared pointer)
```cpp
rl::mdl::UrdfFactory urdf_factory;
std::shared_ptr<rl::mdl::Kinematic> kinematics(
	dynamic_cast<rl::mdl::Kinematic*>(
		urdf_factory.create("C:\\Program Files\\RoboticsLibrary\\0.7.0\\MSVC\\14.1\\x64\\share\\rl-0.7.0\\examples\\rlmdl\\robot_model.urdf")
	)
);
```
or this (without shared pointer)
```cpp
rl::mdl::UrdfFactory urdf_factory;
rl::mdl::Kinematic* kinematics = dynamic_cast<rl::mdl::Kinematic*>(
	urdf_factory.create("C:\\Program Files\\RoboticsLibrary\\0.7.0\\MSVC\\14.1\\x64\\share\\rl-0.7.0\\examples\\rlmdl\\robot_model.urdf")
);
```

Getting the third-party libraries on Windows currently is more work than on Linux or macOS. Building them yourself however should be easier with the CMake `ExternalProject` definitions in the 3rdparty subfolder. Did you have a look at our [tutorial](https://www.roboticslibrary.org/tutorials/build-windows/#dependencies) on this? Microsoft's vcpkg may be an alternative in the future, but at the moment the related Coin and SoQt PR is not merged yet and you still have to compile everything locally.

I know we should do another release with an updated installer at some point, but I'm kind of hoping to get the GitHub Actions for Windows running first in order to automate this.

The third-party binaries in the 0.7 installer should still be compatible with the current master branch. Let me know if you need any specific binaries, I've built all of them locally with VS2019 x64.

Tirine, 2020-10-14:
Hello! I have a URDF file from Kinova that I wanted to use, [it can be found here](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_description/arms/gen3/7dof/urdf/GEN3_URDF_V12.urdf), but it doesnt seem to work. If I try to get info from model, like model->getDof() or anything else, it seems like it's not initialized. 

```c++
int main(int argc, char **argv) {
	try
	{
		rl::mdl::UrdfFactory factory;
		std::shared_ptr<rl::mdl::Model> model;
		model.reset(factory.create("E:\\Home\\User\\Documents\\KORTEX\\master\\kortex-master\\api_cpp\\examples\\GEN3_URDF_V12.urdf"));
		std::cerr << "Model created, DoF: " << model.get()->getDof() << std::endl;
		std::shared_ptr<rl::mdl::Kinematic> kinematic = dynamic_pointer_cast<rl::mdl::Kinematic>(model);
		rl::math::Vector q(7);
		std::cerr << "Vec Loaded " << std::endl;
		q << 10, 10, -20, 30, 50, -10, 1;
		std::cerr << "Vec Set" << q.transpose() << std::endl;
		q *= rl::math::DEG2RAD;
		std::cerr << "Vec in RAD" << q.transpose() << std::endl;
		kinematic->setPosition(q);
		std::cerr << "Pos Set" << std::endl;
		kinematic->forwardPosition();
		std::cerr << "Fwd done" << std::endl;
		rl::math::Transform t = kinematic->getOperationalPosition(0);
		std::cerr << "Transform" << std::endl;
		rl::math::Vector3 position = t.translation();
		std::cerr << "Translate" << std::endl;
		rl::math::Vector3 orientation = t.rotation().eulerAngles(2, 1, 0).reverse();
		std::cout << "Joint configuration in degrees: " << q.transpose() * rl::math::RAD2DEG << std::endl;
		std::cout << "End-effector position: [m] " << position.transpose() << " orientation [deg] " << orientation.transpose() * rl::math::RAD2DEG << std::endl;
	}
	catch(const std::exception& e)
	{
		std::cerr << e.what() << '\n';
	}
	
	return 0;
}
```
gives me this output:

```
GEN3_URDF_V12
link: 0
        cog: -0.000648 -0.000166  0.084487
        inertia: 0.004622 0.004495 0.002079    9e-06    6e-05    9e-06
        mass: 1.697
        name: base_link
link: 1
        cog:  -2.3e-05 -0.010364  -0.07336
        inertia:  0.00457 0.004831 0.001409 0.000448    2e-06    1e-06
        mass: 1.3773
        name: Shoulder_Link
link: 2
        cog:  -4.4e-05  -0.09958 -0.013278
        inertia:  0.011088  0.001072  0.011255 -0.000691         0     5e-06
        mass: 1.1636
        name: HalfArm1_Link
link: 3
        cog:  -4.4e-05 -0.006641 -0.117892
        inertia: 0.010932 0.011127 0.001043 0.000606   -7e-06        0
        mass: 1.1636
        name: HalfArm2_Link
link: 4
        cog:  -1.8e-05 -0.075478 -0.015006
        inertia: 0.008147 0.000631 0.008316  -0.0005        0   -1e-06
        mass: 0.9302
        name: ForeArm_Link
link: 5
        cog:     1e-06 -0.009432 -0.063883
        inertia: 0.001596 0.001607 0.000399 0.000256        0        0
        mass: 0.6781
[... 80 lines cut ...]
Vec Set 10  10 -20  30  50 -10   1
Vec in RAD 0.174533  0.174533 -0.349066  0.523599  0.872665 -0.174533 0.0174533
Pos Set
Fwd done
Transform
Translate
Joint configuration in degrees:  10  10 -20  30  50 -10   1
End-effector position: [m] 0 0 0 orientation [deg]  0 -0  0
```

Meanwhile if I switch to XmlFactory and use one of the XMLs provided in RL, it works and gives me this:
```
Model created, DoF: 6
Vec Loaded 
Vec Set 10  10 -20  30  50 -10   1
Vec in RAD 0.174533  0.174533 -0.349066  0.523599  0.872665 -0.174533 0.0174533
Pos Set
Fwd done
Transform
Translate
Joint configuration in degrees:  10  10 -20  30  50 -10   1
End-effector position: [m] 0.321872  0.20389  1.03928 orientation [deg] -14.9453  39.2476  22.3873
```

Considering that UrdfFactory prints out all the data that it gets, but the model seem to be uninitialized, I would assume the print out is supposed to give some information about that something is missing, but what? I've also tried to use regular pointers instead of the std::shared_ptr and all sorts of other ways, but end up with the same result.

Tirine, 2020-10-15:
I solved it by making an XML based on the data in the URDF and the documentations from [Kinova (User Guide)](https://www.kinovarobotics.com/en/resources/gen3-technical-resources), so if ya wanna add it to your examples:
<details><summary>GEN3_XML_V12.xml</summary>
<i>Tried to add this as a file, but GitHub only accepted TXT etc..</i>
<p>

```XML
<?xml version="1.0" encoding="UTF-8"?>
<rlmdl xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="rlmdl.xsd">
	<model>
		<manufacturer>Kinova</manufacturer>
		<name>Gen3 7-dof</name>
		<world id="world">
			<rotation>
				<x>0</x>
				<y>0</y>
				<z>0</z>
			</rotation>
			<translation>
				<x>0</x>
				<y>0</y>
				<z>0</z>
			</translation>
			<g>
				<x>0</x>
				<y>0</y>
				<z>9.81</z>
			</g>
		</world>
		<frame id="frame_base"/>
		<body id="base_link">
			<ignore idref="Shoulder_Link"/>
			<cm>
				<x>0.000648</x>
				<y>-0.000166</y>
				<z>0.084487</z>
			</cm>
			<i>
				<xx>0.004622</xx>
				<xy>0.000009</xy>
				<xz>0.00006</xz>
				<yy>0.004495</yy>
				<yz>0.000009</yz>
				<zz>0.002079</zz>
			</i>
			<m>1.697</m>
		</body>
		<frame id="frame0"/>
		<frame id="frame1"/>
		<body id="Shoulder_Link">
			<ignore idref="base_link"/>
			<ignore idref="HalfArm1_Link"/>
			<cm>
				<x>0.000023</x>
				<y>-0.010364</y>
				<z>-0.07336</z>
				</cm>
			<i>
				<xx>0.00457</xx>
				<xy>0.000001</xy>
				<xz>0.000002</xz>
				<yy>0.004831</yy>
				<yz>0.000448</yz>
				<zz>0.001409</zz>
			</i>
			<m>1.3773</m>
		</body>
		<frame id="frame2"/>
		<frame id="frame3"/>
		<body id="HalfArm1_Link">
			<ignore idref="Shoulder_Link"/>
[... 500 lines cut ...]
		</fixed>
		
		
		<!-- Only needed if we add an end effector <frame id="frame14"/>
		<body id="EndEffector_Link"/> -->
		<!-- <revolute id="EndEffector">  -->
		<!-- type="fixed"> -->
		<!-- <frame>
				<a idref="Bracelet_Link"/>
				<b idref="EndEffector_Link"/>
		</frame>
			<axis>
				<x>0</x>
				<y>0</y>
				<z>0</z>
			</axis>
		</revolute> -->
		<!-- <origin xyz="0 0 -0.0615250000000001" rpy="3.14159265358979 1.09937075168372E-32 0"/> -->
	</model>
</rlmdl>
```
</p>
</details>

I only needed it for kinematics calcs, so I've only checked the DH parameters, probably loads of things not correct elsewhere... Compared to the documentation in the User Guide, they assumed a worldframe with Z pointing downwards, so I accounted for that with just inverting the Z translation from the DH d param instead on `<fixed id="fixed_base">` .

rickertm, 2020-10-17:
There was an issue in `rl::mdl::UrdfFactory` in 0.7.0 when parsing continuous joints that has been fixed in 9bcf3e7eb05deeb7d998aa3a2d10891962003031. The current version on the master branch should work. This also includes a `rl::sg::UrdfFactory` for parsing the geometry information in the URDF file. After fixing the mesh filenames to relative paths and the correct files, you can use this in rlCoachMdl:
[image]
I'll have a look into adding a full model for the Gen3 Kinova arm including geometry if there are no more issues with the Gen2 version, as this seems to be interesting for some users.

MarceloEO, 2024-05-17:
Dear @rickertm and RL community,

I am grateful for all the valuable work you have put into RL. I recently discovered your library a few days ago, and as a newcomer, I am still finding my way around.

I'm having trouble replicating the results shown in your previous screenshot using rlCoachMdl. Could you please provide more details on your Kinova SCENEFILE and MODELFILE?

I am using the URDF file from the following URL: https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_description/arms/gen3/6dof/urdf with the corrected paths for the meshes.

rl::mdl::UrdfFactory urdf_factory;

rl::mdl::Model *model = dynamic_cast<rl::mdl::Model *>(
		urdf_factory.create("GEN3-6DOF_VISION_URDF_ARM_V01.urdf"));

rl::mdl::Kinematic *kinematic = dynamic_cast<rl::mdl::Kinematic *>(
		urdf_factory.create("GEN3-6DOF_VISION_URDF_ARM_V01.urdf"));

Thank you for your attention and assistance.

Best regards,