roboticslibrary/rl #18: building RL lib with VISP library and openCV

engr-zubair, 2020-05-08:
hi dear, I am working on visual servo control ,therefore I am using VISP library for visual servoying part and I wish to utilize RL LIBRARY for other robot operations and path planning deployment.
I have successfully built visp+ rl library thru CMake 3.4 for VS-2017 c++ using win dev-kit 10.17 , the rl lib and visp core functions are workingtogether  properly..also both libraries are working well individually . 
the issue I am facing is...whenever I add header files or call OpenCV/visp GDI display and vision functionality I got the repeated errors as below , there is some functions overlapping error between these two libraries. can you have a look at this and guide me through solution.

Error (active)	E0311	cannot overload functions distinguished by return type alone	visp-rl-jac-kinova-cart	C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\um\WinSock2.h	2197	

Error (active)	E0040	expected an identifier	visp-rl-jac-kinova-cart	C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\shared\ws2def.h	437	

Error (active)	E0338	more than one instance of overloaded function "connect" has 'C' linkage	visp-rl-jac-kinova-cart	C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\um\WinSock2.h	1685	

repeated for " accept, bind ,getpeername , getsockname " and so on...
looks like these functions are defined else where with same name either in windows development kit files or in visp files...

please guide , thanks.
below is my CmakeLists file

`project(visp-rl-kin-jac)
cmake_minimum_required(VERSION 2.6)
set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
add_definitions(-D_USE_MATH_DEFINES)
find_package(VISP REQUIRED)
find_package(RL COMPONENTS MDL KIN MATH PLAN REQUIRED )

set(example_cpp
  visp-rl-jac-kinova-joint.cpp
  visp-rl-jac-kinova-cart.cpp
)

if(MSVC AND CMAKE_CXX_COMPILER_VERSION VERSION_LESS 19.00.23918)
	target_compile_definitions(
		visp-rl-kin-jac
		PUBLIC
		BOOST_ALL_NO_LIB
		BOOST_CHRONO_HEADER_ONLY
		BOOST_ERROR_CODE_HEADER_ONLY
		BOOST_SYSTEM_NO_DEPRECATED
	)
endif()

foreach(cpp ${example_cpp})
  visp_add_target(${cpp})
  if(COMMAND visp_add_dependency)
    visp_add_dependency(${cpp} "examples")
  endif()
endforeach()
target_link_libraries(visp-rl-jac-kinova-joint ${RL_LIBRARIES})
target_link_libraries(visp-rl-jac-kinova-cart ${RL_LIBRARIES})`

rickertm, 2020-05-09:
Without a detailed log I have to guess, but a common error is mixing winsock and winsock2, the latter is used by rl::hal::Socket. In order to solve this you have to either include `winsock2.h` before including `windows.h` or add `WIN32_LEAN_AND_MEAN` to you compile definitions.

VISP does something similar, e.g., in [vpWin32API.h](https://github.com/lagadic/visp/blob/master/modules/gui/include/visp3/gui/vpWin32API.h#L48-L50).

engr-zubair, 2020-05-10:
Thankyou very much...this really solves the problem, adding `winsock2.h` in start of headers , now it is building error free.
I really appreciate your help !

closed 2020-05-10 by engr-zubair