roboticslibrary/rl #19: how to get data from rl::math::matrix

engr-zubair, 2020-05-16:
hi dear, I have implemented and calculated a robot jacobian using rl-library kinematics functions using xml model, as I am using a visual servo control law form VISP library , I am unable to use this jacobian matrix obtained in control law calculation , 
I also cannot extract individual data elements and store them in another array or matrix as double value. so I can built-from this data a matrix which I can use in control law jacobian. 

kindly share code how to extract or make a matrix / 2Darray of double values from rl-math-kinematics jacobian matrix.
note: both visp and rl libraries use Eigen for solving math..

`const rl::math::Matrix& jac = kinematics->getJacobian();`
vpMatrix jacob; 'a matrix class using eigen for VISP library' 
visp matrix lib ref [visp-matrix class](https://visp-doc.inria.fr/doxygen/visp-daily/group__group__core__matrices.html)

how to assign convert them in each other, using core functionality of c++ and eigen ..??
thanks!

rickertm, 2020-05-16:
When VISP is compiled with support for Eigen (`VISP_HAVE_EIGEN3`), it apparently includes the functions [eigen2visp](https://visp-doc.inria.fr/doxygen/visp-daily/namespacevp.html#a4c1ad2859d8a32bff29e151e19055533) and [visp2eigen](https://visp-doc.inria.fr/doxygen/visp-daily/namespacevp.html#a0a0889c768989f9397cbeb537495ef54) for converting between the two structures in `visp3/core/vpEigenConversion.h`.

Here's a short example on how to use this:
```cpp
#include <iostream>
#include <rl/math/Matrix.h>
#include <visp3/core/vpEigenConversion.h>
#include <visp3/core/vpMatrix.h>

int
main(int argc, char** argv)
{
	rl::math::Matrix m1(2, 3);
	m1.setRandom();
	std::cout << "RL" << std::endl << m1 << std::endl;
	
	vpMatrix m2;
	vp::eigen2visp(m1, m2);
	std::cout << "VISP" << std::endl << m2 << std::endl;
	
	rl::math::Matrix m3;
	vp::visp2eigen(m2, m3);
	std::cout << "RL" << std::endl << m3 << std::endl;
	
	return EXIT_SUCCESS;
}
```

engr-zubair, 2020-05-18:
thanks It was really helpful, it does the job .

closed 2020-05-18 by engr-zubair