roboticslibrary/rl #20: Website syntax conflicts

BeardedVagabond, 2020-05-29:
I've noticed a few discrepancies on the website between 0.7.0 syntax and master branch syntax. I may have made mistakes in my implementation, but was curious about the differences.

api/ rl::sg section
- `scene.load("scene.xml")`
  - This is only available in 0.7.0 and an XMLFactory must be used in the master branch
  - Not very serious assuming the api page is dealing only with the stable 0.7.0 release, but maybe a small note would help anyone that built from master

api/ rl::plan section
- `if (rlmdl) model.mdl = kinematics`
  - This appears impossible in 0.7.0 as the model.mdl is a pointer to `rl::mdl::Dynamic*` and not `rl::mdl::Kinematic*`
  - Is it possible to use mdl kinematics in c++ planners on the 0.7.0 branch? On the master branch it works extremely well. I never tested with xml descriptions as I prefer to build the planner in c++

tutorials/build-windows/ page
- I've noticed that running the cmake commands requires administrator privileges if installing to `C:\Program Files\`, however, this is not mentioned on the page. Installing 3rdparty dependencies failed on access privileges for me when not in an administrator terminal.

rickertm, 2020-05-29:
There are indeed a few differences between the 0.7 and the master branch. The documentation on the website currently describes the 07 branch.  The plan is to update the documentation in the future to be able to switch between different releases, but this may take a while.

>     * `scene.load("scene.xml")`
>       
>       * This is only available in 0.7.0 and an XMLFactory must be used in the master branch
>       * Not very serious assuming the api page is dealing only with the stable 0.7.0 release, but maybe a small note would help anyone that built from master

As this seems to be a common question I've now added a short note to the page with an example of the updated API for loading a scene description.

> * `if (rlmdl) model.mdl = kinematics`
>
>       * This appears impossible in 0.7.0 as the model.mdl is a pointer to `rl::mdl::Dynamic*` and not `rl::mdl::Kinematic*`
>       * Is it possible to use [...]

You can use mdl kinematics in planners on the 0.7 branch, you just have to create a `rl::mdl::Dynamic` class when constructing a model. The `rl::mdl::XmlFactory` does this by default and the only difference are a added functions for dynamic calculations and member variables for storing the mass matrix, centrifugal and Coriolis vector, as well as the gravity vector. As these are currently not needed in the path planning calculations, I've removed this restriction on the master branch.

>     * I've noticed that running the cmake commands requires administrator privileges if installing to `C:\Program Files\`, however, this is not mentioned on the page. Installing 3rdparty dependencies failed on access privileges for me when not in an administrator terminal.

Could you please elaborate on this? I've just tested this again on my local installation (installed to `C:\Program Files\`) and did not require admin privileges when building a demo program. The installer allows specifying a different location when the user does not have access to system directories, but after a successful installation to `C:\Program Files\` this should work as well.

BeardedVagabond, 2020-06-08:
Apologies for taking so long to reply here. 

I noticed the clarification showing `if (rlmdl) model.mdl = dynamic; // for rl::mdl models`. Thanks for your note about how to achieve this in 0.7.0 branch. 

The note on scene loading is appreciated as well

I'm honestly not 100% sure why I can't seem to install unless I'm in an admin command window. That could potentially be a user configuration thing on my end. In any case, I feel like if someone is building a repo from source they can probably troubleshoota ccess privledges well enough that it isn't an issue :) 

I'll close this issue now. Thanks again for the library and your always quick responses!

closed 2020-06-08 by BeardedVagabond