roboticslibrary/rl #21: rl::sg::fcl::Scene::addCollisionObject

ycheng-l, 2020-06-24:
Can you give me some demos about 'rl::sg::fcl::Scene::addCollisionObject'? I want to use 'octomap::OcTree' check robot collision. Thanks~~~

rickertm, 2020-06-25:
This function is currently only used internally for adding objects to the FCL scene's broadphase manager. The `fcl::CollisionObject` is actually stored inside `rl::sg::fcl::Shape`. In order to use an `octomap::OcTree`, a `fcl::OcTree` geometry would have to be used when creating the object in the shape's constructor. e.g., via a modified constructor or a subclass.

https://github.com/roboticslibrary/rl/blob/22a051131c45e3e02e47837e4b71260ea0710f45/src/rl/sg/fcl/Shape.cpp#L235

ycheng-l, 2020-06-26:
Thanks for your reply! Actually i want adding real time point cloud data in planning scene, like this: https://www.youtube.com/watch?v=GZwSBIBKRAA&list=PLcZc0GY2V0Ew7jQ9RGU8dVtY_KZ77uuY_&index=14
I can't find API add real time data. Maybe you can give me some suggestions to add real time point cloud data？

rickertm, 2020-06-27:
I have added an optional constructor for `rl::sg::fcl::Shape` in acf0ee62145a3bb077dad10e0f22bfd5203e1cba that accepts a `fcl::CollisionGeometry`. With this, you can now add a `fcl::OcTree` geometry around a `octomap::OcTree` to update your scene with point cloud data.

```cpp
std::shared_ptr<octomap::OcTree> octree = std::make_shared<octomap::OcTree>(0.1);
std::shared_ptr<fcl::OcTree<rl::math::Real>> octreeGeometry = std::make_shared<fcl::OcTree<rl::math::Real>>(octree);
rl::sg::fcl::Model* model = dynamic_cast<rl::sg::fcl::Model*>(scene->create());
rl::sg::fcl::Body* body = dynamic_cast<rl::sg::fcl::Body*>(model->create());
rl::sg::fcl::Shape* shape = new rl::sg::fcl::Shape(octreeGeometry, body);

for (std::size_t i = 0; i < 100; ++i)
{
	rl::math::Vector3 random = rl::math::Vector3::Random();
	octree->updateNode(octomap::point3d(random.x(), random.y(), std::abs(random.z())), true);
}
```
Here's a screenshot of a quick example that combines this with rlPlanDemo. You can use `octreeGeometry->toBoxes()` to generate a list of boxes for a 3D visualization of the octree, e.g., using `SoVRMLBox`.

[image]

ycheng-l, 2020-06-28:
Thank you for sharing the code! I have solved the problem now.

ycheng-l, 2020-08-05:
hi~~~i have a new question:
for example, i use **rl::sg::fcl::Scene** load() function load my robot(have end-effector) ,then i use **rl::plan::RrtConCon** planner success plan a collision free path.But when i loaded a no end-effector robot,then
use rl::sg::fcl::Scene->getModel()->add() function added my end-effector Body,the planner can not check end-effector collision. How to solve this problem? ^_^

rickertm, 2020-08-06:
The planner requires both a kinematic model and a geometry in order to update the individual links for each sampled configuration. If an additional body, e.g., a tool, is only added to the geometry model but not to its kinematic model, the planner is not able to update the respective frames accordingly.

You can either include the additional body for the end effector in the kinematic model of the robot or you can add the tool as a shape with a static transform to the end-effector's body. The latter will not require you to update the kinematic model.

ycheng-l, 2020-08-06:
My kinematic model has 8 bodies, it includes end-effector. 
I want to know how to add end-effector model?(not load once, because I have to end-effectors)

closed 2020-10-27 by ycheng-l

marcoojer, 2021-07-02:
Can you please provide, the code for the visualization of the octomap?

rickertm, 2021-07-05:
Here's the example code I used for this that can be added to rlPlanDemo, e.g., at the end of `MainWindow::load`. It looks for all `fcl::OcTree` shapes in the FCL collision scene and adds them to the Coin3D visualization scene.
```cpp
if (rl::sg::fcl::Scene* scene = dynamic_cast<rl::sg::fcl::Scene*>(this->scene.get()))
{
	for (std::size_t i = 0; i < scene->getNumModels(); ++i)
	{
		for (std::size_t j = 0; j < scene->getModel(i)->getNumBodies(); ++j)
		{
			for (std::size_t k = 0; k < scene->getModel(i)->getBody(j)->getNumShapes(); ++k)
			{
				if (rl::sg::fcl::Shape* shape = dynamic_cast<rl::sg::fcl::Shape*>(scene->getModel(i)->getBody(j)->getShape(k)))
				{
					if (const fcl::OcTree<rl::math::Real>* octree = dynamic_cast<const fcl::OcTree<rl::math::Real>*>(shape->getCollisionObject()->collisionGeometry().get()))
					{
						std::vector<std::array<rl::math::Real, 6>> boxes = octree->toBoxes();
						
						SoVRMLTransform* vrmlRoot = new SoVRMLTransform();
						
						rl::math::Transform frame;
						scene->getModel(i)->getBody(j)->getFrame(frame);
						rl::math::Transform transform;
						shape->getTransform(transform);
						
						rl::math::Transform frameTransform = frame * transform;
						
						SbMatrix matrix;
						
						for (int m = 0; m < 4; ++m)
						{
							for (int n = 0; n < 4; ++n)
							{
								matrix[m][n] = static_cast<float>(frameTransform(n, m));
							}
						}
						
						vrmlRoot->setMatrix(matrix);
						
						for (std::size_t l = 0; l < boxes.size(); ++l)
						{
							SoVRMLShape* vrmlShape = new SoVRMLShape();
							SoVRMLBox* vrmlBox = new SoVRMLBox();
							vrmlBox->size.setValue(boxes[l][3], boxes[l][3], boxes[l][3]);
							vrmlShape->geometry = vrmlBox;
							SoVRMLMaterial* vrmlMaterial = new SoVRMLMaterial();
							vrmlMaterial->diffuseColor.setValue(1.0 - boxes[l][4], 1.0 - boxes[l][4], 1.0 - boxes[l][4]);
							vrmlMaterial->emissiveColor.setValue(1.0 - boxes[l][4], 1.0 - boxes[l][4], 1.0 - boxes[l][4]);
							SoVRMLAppearance* vrmlAppearance = new SoVRMLAppearance();
							vrmlAppearance->material = vrmlMaterial;
							vrmlShape->appearance = vrmlAppearance;
							SoVRMLTransform* vrmlTransform = new SoVRMLTransform();
							vrmlTransform->translation.setValue(boxes[l][0], boxes[l][1], boxes[l][2]);
							vrmlTransform->addChild(vrmlShape);
							vrmlRoot->addChild(vrmlTransform);
						}
						
						this->scene2->root->addChild(vrmlRoot);
					}
				}
			}
		}
	}
}
```