roboticslibrary/rl #22: [Question] Element position after reversing multiplication order within XmlFactory.cpp

DmytroJK, 2020-06-26:
Hello, 

I currently using HOOPS Visualize together with Robotics Library to implement motion planning and collision detection. Everything was working fine until I used several angles about different axes. Through rlCoachMdl.exe I could see that position of elements with this transformation is broken. 
Within Visualize I rotate an element in this way:
```
segmentKey.ShowModellingMatrix(out MatrixKit currentMatrix);

MatrixKit rotatedMatrixAboutX = new MatrixKit().Rotate(angleX, 0, 0);
rotatedMatrixAboutX.MultiplyAndAssign(currentMatrix);

MatrixKit rotatedMatrixAboutY = new MatrixKit().Rotate(0, angleY, 0);
rotatedMatrixAboutY.MultiplyAndAssign(rotatedMatrixAboutX);

MatrixKit rotatedMatrixAboutZ = new MatrixKit().Rotate(0, 0, angleZ);
rotatedMatrixAboutZ.MultiplyAndAssign(rotatedMatrixAboutY);
```
Considering that MultiplyAndAssign() multiplies the parameter matrix as a right operand, I assumed that the multiplication order is ZYX. And it corresponds multiplication order of the "**fixed**" element within **XmlFactory.cpp** (line 308). 

But while comparing transformations I realized that the problem may really be in the multiplication order. So I reverted it in this way: 
```

                f->t = ::rl::math::AngleAxis(
				path.eval("number(rotation/x)").getValue< ::rl::math::Real>(0) * ::rl::math::DEG2RAD,
				::rl::math::Vector3::UnitX()
		) * ::rl::math::AngleAxis(
				path.eval("number(rotation/y)").getValue< ::rl::math::Real>(0) * ::rl::math::DEG2RAD,
				::rl::math::Vector3::UnitY()
		) * ::rl::math::AngleAxis(
				path.eval("number(rotation/z)").getValue< ::rl::math::Real>(0) * ::rl::math::DEG2RAD,
				::rl::math::Vector3::UnitZ()
		);

```
And after that matrixes values start to be equal in array representation (matrix from Visualize and matrix from RL). But after quick testing, I understood that rotation about the Z-axis is completely ignored. XY rotation works perfect and elements position correspond within two scenes in this case. For me, it is very strange, because if we have "**home**" value about Z-axis it works fine.

Could you please give me a hint about what can cause this behavior? And what I must change to get the right orientation?

_Additional information:
I think that matrix order within Visualize is that:
1 0 0 0
0 1 0 0
0 0 1 0
X Y Z 1
where X,Y,Z - translation. So array values go as a row._

kinematics0.xml
```
    <fixed id="fixed4">
      <frame>
        <a idref="body1" />
        <b idref="frame3" />
      </frame>
      <rotation>
        <x>69</x>
        <y>34</y>
        <z>74</z>
      </rotation>
      <translation>
        <x>130</x>
        <y>0</y>
        <z>251</z>
      </translation>
    </fixed>
    <revolute id="joint2">
      <frame>
        <a idref="frame3" />
        <b idref="frame4" />
      </frame>
      <axis>
        <x>0</x>
        <y>0</y>
        <z>1</z>
      </axis>
      <max>123</max>
      <min>-98</min>
      <speed>180</speed>
    </revolute>

    <home>
      <q unit="deg">25</q>
    </home>
```
rotation z is ignored

Thank you
Very grateful for this project

rickertm, 2020-06-26:
The order of multiplication in `XmlFactory.cpp` is correct for Euler-ZYX angles. Please check what kind of coordinate system your setup requires. RL uses Eigen3 for its matrix calculations. The [`Eigen::AngleAxis`](https://eigen.tuxfamily.org/dox/classEigen_1_1AngleAxis.html) rotations use a right-handed coordinate system when converting to a 3x3 rotation matrix, resulting in the following matrices (c.f., [basic rotations](https://en.wikipedia.org/wiki/Rotation_matrix#Basic_rotations) and [rotation matrix](https://mathworld.wolfram.com/RotationMatrix.html)):
```
Eigen::AngleAxisd(EIGEN_PI, Eigen::Vector3d::UnitX).toRotationMatrix() ->
 1  0  0
 0 -1  0
 0  0 -1

Eigen::AngleAxisd(EIGEN_PI, Eigen::Vector3d::UnitY).toRotationMatrix() ->
-1  0  0
 0  1  0
 0  0 -1

Eigen::AngleAxisd(EIGEN_PI, Eigen::Vector3d::UnitZ).toRotationMatrix() ->
-1  0  0
 0 -1  0
 0  0  1
```
By default, Eigen uses a [row-major order](https://eigen.tuxfamily.org/dox/group__TopicStorageOrders.html) for storing the matrix. An [`Eigen::Transform`](https://eigen.tuxfamily.org/dox/classEigen_1_1Transform.html) combines rotation and translation and can be converted to a 4x4 matrix.
```
Eigen::Affine3d(Eigen::Translation3d(1, 2, 3)).matrix() ->
1 0 0 x
0 1 0 y
0 0 1 z
0 0 0 1
```

Your example for the translation in Visualize suggests it might be expecting a left-handed coordinate system, c.f., [translation](https://docs.microsoft.com/en-us/windows/win32/direct3d9/transforms#translate) and [rotation](https://docs.microsoft.com/en-us/windows/win32/direct3d9/transforms#rotate).

DmytroJK, 2020-06-30:
Thank you for the quick reply.

>Please check what kind of coordinate system your setup requires

Visualize uses a right-handed coordinate system as well and Z-axis points up.
[image]

The position of every element within the scene is based on the position of its parent element. So all transformation that I use is applying to the local modeling matrix.

>Your example for the translation in Visualize suggests it might be expecting a left-handed coordinate system

I know what you mean and had already checked this option. If the problem is really because of this, then XY rotation (applying rotation about X-axis and Y-axis) would be incorrect because of matrix order. And in this case arrays representations of modeling matrixes would be different.

Steps of my scenario:
1. Implement all three rotation to an element within Visualize by multiplying appropriate matrixes 
(as I mentioned MatrixKit rotatedMatrixAboutX = new MatrixKit().Rotate(angleX, 0, 0);)

The same angles I can get by these formulas: 
```
X = -Math.Atan2(elements[9], elements[10]) * (180 / Math.PI));
Y = -Math.Asin(-elements[8]) * (180 / Math.PI);
Z = -Math.Atan2(elements[4], elements[0]) * (180 / Math.PI);
```
where `elements` is the array representation of the modeling matrix

2. Write the values of the angles in **kinematics0.xml**
3. Within **XmlFactory.cpp** I multiplying angles in reversal order XYZ and get the same array representation as in Visualize.
4. The element gets into the same position but without rotation about Z-Axis.

Maybe there are some restrictions about Z-axis within **PlueckerTransform** since we apply the rotation values there too?

rickertm, 2020-06-30:
On the current master branch, `PlueckerTransform` internally uses a `Transform` and should not add any additional restrictions. Please note that this was modified from the 0.7 branch.

In order to narrow this down and without access to HOOPS Visualize, could you please provide some additional information? The full values of the individual matrices in your multiplications as well as the final result would be very helpful.

Here are the values for your rotation example (69°, 34°, 74°) using Eigen3, first the individual matrices
```
Eigen::AngleAxisd X(69.0 / 180.0 * EIGEN_PI, Eigen::Vector3d::UnitX());
X.toRotationMatrix() ->
1 0         0
0 0.358368 -0.93358
0 0.93358   0.358368
```
```
Eigen::AngleAxisd Y(34.0 / 180.0 * EIGEN_PI, Eigen::Vector3d::UnitY());
Y.toRotationMatrix() ->
 0.829038 0  0.559193
 0        1  0
-0.559193 0  0.829038
```
```
Eigen::AngleAxisd Z(74.0 / 180.0 * EIGEN_PI, Eigen::Vector3d::UnitZ());
Z.toRotationMatrix() ->
0.275637 -0.961262 0
0.961262  0.275637 0
0         0        1
```
and the combined matrix in ZYX multiplication order
```
Eigen::Matrix3d ZYX = (Z * Y * X).toRotationMatrix();
 0.228514 -0.200588  0.952652
 0.796922  0.600608 -0.0646959
-0.559193  0.773973  0.2971
```
This matches the `Eigen::EulerAnglesZYX` calculation
```
Eigen::EulerAnglesZYXd eulerZYX(
  74.0 / 180.0 * EIGEN_PI,
  34.0 / 180.0 * EIGEN_PI,
  69.0 / 180.0 * EIGEN_PI
);
eulerZYX.toRotationMatrix() ->
 0.228514 -0.200588  0.952652
 0.796922  0.600608 -0.0646959
-0.559193  0.773973  0.2971
```
Based on the symbolic formula
```
 cos(z) cos(y)   -sin(z) cos(x) + cos(z) sin(y) sin(x)    sin(z) sin(x) + cos(z) sin(y) cos(x)
 sin(z) cos(y)    cos(z) cos(x) + sin(z) sin(y) sin(x)   -cos(z) sin(x) + sin(z) sin(y) cos(x)
-sin(y)           cos(y) sin(x)                           cos(y) cos(x)
```
the angles can be extracted as follows (assuming no [Gimbal lock](https://en.wikipedia.org/wiki/Gimbal_lock))
```
x =  atan2(ZYX(2, 1), ZYX(2, 2)) * (180.0 / EIGEN_PI);
y = -asin(ZYX(2, 0)) * (180.0 / EIGEN_PI);
z =  atan2(ZYX(1, 0), ZYX(0, 0)) * (180.0 / EIGEN_PI);
x: 69
y: 34
z: 74
```
<hr>

For the multiplication order XYZ this looks quite different of course
```
Eigen::Matrix3d XYZ = (X * Y * Z).toRotationMatrix();
0.228514 -0.796922  0.559193
0.488382 -0.403049 -0.773973
0.842178  0.449963  0.2971
```
and matches `Eigen::EulerAnglesXYZ`
```
Eigen::EulerAnglesXYZd eulerXYZ(
  69.0 / 180.0 * EIGEN_PI,
  34.0 / 180.0 * EIGEN_PI,
  74.0 / 180.0 * EIGEN_PI
);
0.228514 -0.796922  0.559193
0.488382 -0.403049 -0.773973
0.842178  0.449963  0.2971
```
Based on the symbolic formula
```
 cos(z) cos(y)                          -sin(z) cos(y)                             sin(y)
 cos(z) sin(y) sin(x) + sin(z) cos(x)   -sin(z) sin(y) sin(x) + cos(z) cos(x)     -cos(y) sin(x)
-cos(z) sin(y) cos(x) + sin(z) sin(x)    sin(z) sin(y) cos(x) + cos(z) sin(x)      cos(y) cos(x)
```
the angles are calculated as (assuming no [Gimbal lock](https://en.wikipedia.org/wiki/Gimbal_lock))
```
x = -atan2(XYZ(1, 2), XYZ(2, 2)) * (180.0 / EIGEN_PI);
y =  asin(XYZ(0, 2)) * (180.0 / EIGEN_PI);
z = -atan2(XYZ(0, 1), XYZ(0, 0)) * (180.0 / EIGEN_PI);
x: 69
y: 34
z: 74
```