roboticslibrary/rl #24: [question]. Compile without CMAKE.

santdiego, 2020-07-03:
Hi. Is it possible to compile rl programs using an IDE instead of cmake? I usually work in QT Creator with qmake, what would be the dependencies that should be included / linked in the project?
Great job ! Thank you!

rickertm, 2020-07-04:
It is definitely possible to build programs and link against the RL without using CMake, but you will have to manually specify the required libraries, include directories, and compiler definitions. An overview of the connections between [RL components](https://www.roboticslibrary.org/dependencies.svg) and their [third-party dependencies](https://www.roboticslibrary.org/external-dependencies.svg) can be found on the website. The actual dependencies and settings used in an installation can be found in the `lib/cmake/rl-*/rl-export*.cmake` files.

If you are using the Windows installer, the required includes should all be in the folder `include`, but you also have to add the `rl-0.7.0` and `eigen3` subfolders. All libraries can be found under `lib`, the required ones depend on the features you are using, see the images above. Compiler definitions include `BOOST_ALL_NO_LIB` for Boost, `COIN_DLL` and `SOQT_DLL` for Coin3D, and `dDOUBLE` or `dSINGLE` depending on the float precision used in the ODE library.

Several IDEs can natively use CMake files, e.g., MSVC, CLion, and Qt Creator. I typically use Qt Creator for this and it works quite well. For several others you can also use CMake to create IDE-specific project files. A current list can be found [here](https://cmake.org/cmake/help/latest/manual/cmake-generators.7.html#ide-build-tool-generators).

santdiego, 2020-07-10:
I am compiling in vscode with cmake and it works perfect. Thank you very much.

closed 2020-07-10 by santdiego