roboticslibrary/rl #25: 'rl::mdl::XmlFactory' memory leak?

ycheng-l, 2020-07-15:
hello~ ^_^
'rl::mdl::XmlFactory' memory leak?

`while(1)`
`{`
`rl::mdl::XmlFactory factory;`
`std::shared_ptr<rl::mdl::Model> kinematicModel`
`kinematicModel.reset(factory.create(mdlModelPath));`
`kinematicModel.reset();`
`}`

rickertm, 2020-07-15:
There was an issue with handling allocated stringval objects from XPath string queries that should be fixed by d9b49632ec867b86a6df56a3d609dc18564d2b2d. Let me know if this addresses your issue as well.

Otherwise, please provide more details on this, i.e., a full code example, your build setup (OS, compiler, etc.) and the relevant output of your memory checker. If your memory checker reports about unused memory related to libxml2 or libxslt, please make sure to also use the corresponding [cleanup functions](http://xmlsoft.org/xmlmem.html) of these libraries at the very end of your program:
```cpp
#include <libxml/globals.h>
#include <libxslt/xslt.h>

int
main(int argc, char** argv)
{
	...

	xmlCleanupParser();
	xsltCleanupGlobals();
	return 0;
}
```

ycheng-l, 2020-07-16:
Thanks your reply! 
It's useless replacing new rl/xml/Object.h. `~Object()` not work.
Here is my test code. I use VS2019 and rl-0.7.0 release building.Actually, loop `factory.create("model.xml")`will taking memory increase.

demo1:
```
#include <rl/mdl/XmlFactory.h>
#include <rl/mdl/Model.h>
int main(int argc, char** argv)
{
	while (1)
	{
		rl::mdl::XmlFactory factory;
		std::shared_ptr<rl::mdl::Model> kinematicModel;
		kinematicModel.reset(factory.create("model.xml"));
		kinematicModel.reset();
	}
}
```
demo2:
```
#include <rl/mdl/XmlFactory.h>
#include <rl/mdl/Model.h>
int main(int argc, char** argv)
{
	while (1)
	{
		rl::mdl::XmlFactory factory;
		factory.create("model.xml");
	}
}
```

rickertm, 2020-07-16:
I do not have any memory leaks or segfaults when running your test code with a valid model file on VS2019 and RL version 0.7.0. An output of your debugger would help to narrow down the issue. As you did not include the file `model.xml`, please make sure that it is a valid definition and that it can actually be loaded.

The `rl::mdl::XmlFactory` will throw an exception if the model file has an error. I would recommend adding a `try/catch` block around this part of your code to check for this. The folder `examples/rlmdl` also includes a number of valid example model files you can use for testing.
```cpp
try
{
	while (1)
	{
		rl::mdl::XmlFactory factory;
		std::shared_ptr<rl::mdl::Model> kinematicModel;
		kinematicModel.reset(factory.create("model.xml"));
		kinematicModel.reset();
	}
}
catch (const std::exception& e)
{
	std::cout << e.what() << std::endl;
	return 1;
}
```

ycheng-l, 2020-08-05:
Thanks~
I build latest version,it works well !!!

424358225, 2020-09-05:
rl::sg::so::Scene can load xml file,but rl::mdl::XmlFactory creat() will be crashed.

closed 2020-10-27 by ycheng-l