roboticslibrary/rl #26: UniversalRobots: How to get actual joint-positions when not actively running a trajectory?

Mottevator, 2020-07-23:
Hello,
I am trying to use this library to control a UR10 and have the following problem:

I use rl::hal::UniversalRobotsRtde and open a connection to a UR10. I can run a trajectory on the robot but i cannot get actual joint-positions when i do not actively run a trajectory, after calling the open()-function, the function getJointPositions() return a zero-vector until start() and step() functions are called.

When i finish the trajectory and call the stop() function, the values returned by the function getJointPositions() do not update anymore, so they stay at the last values, even when I move the robot.

I need the actual values of the robot, also when i am not actively controlling the robot (for monitoring, collision-checking etc.) - How can i do this? The robot is always streaming the data at a rate of 125Hz, so the data should be provided.

I also tried implementing the rl::hal::UniversalRobotsRealTime class, here I got the actual values after I executed open(), start() and step(). But after this, the values are not updating when i call step(), so I have to stop, close the connection, reconnect, start and step again to get updated values.

Kind regards!

rickertm, 2020-07-23:
The open() function only opens the socket connection to the robot controller, while the start() function actually initiates the data exchange with the controller. For a cyclic device such as the UR, the step() function has to be called with the specified update rate of the controller in order to send commands and receive updates from the hardware. The get functions can only access the information from the last update, while the commands from all set functions will be sent at the next step() call.

The `rl::hal::UniversalRobotsRtde` class uses the UR's [RTDE protocol](https://www.universal-robots.com/articles/ur-articles/real-time-data-exchange-rtde-guide/) on port 30004 to exchange data and controls the robot via an uploaded script on port 30002 that reads the target information from the RTDE registers.

The `rl::hal::UniversalRobotsRealtime` class only receives information via the 30003 port and does not send any control commands. It still requires calling the step() function at the given intervals.

The `rl::hal::UniversalRobotsDashboard` class can be used to remotely power on/off the system, release the brakes etc. without the use of the teach pendant.

Mottevator, 2020-07-24:
Thank you very much, it works now! 

I use rl::hal::UniversalRobotsRealtime for monitoring the robot state. My mistake was that I did not call the step() function with the proper frequency. So it seems like the received messages get buffered at the receiver if the freuqency of calling step() is too low, which results in delayed robot data.