roboticslibrary/rl #27: use xacro files with robotics library

santdiego, 2020-07-24:
Hi. I need to use a robot model with extension "* .xacro":  https://github.com/ROBOTIS-GIT/ROBOTIS-MANIPULATOR-H/tree/master/manipulator_h_description/urdf
Is it possible to load this model with the Robotics Library?
Thank you!

rickertm, 2020-07-26:
You can use `rl::mdl::UrdfFactory` and `rl::sg::UrdfFactory` for loading kinematics and geometry models from URDF files. If the URDF file uses ROS' xacro XML macro language, you first have to convert the file to a native URDF file
```shell
$ rosrun xacro xacro --inorder -o manipulator_h.urdf manipulator_h.xacro
```
and convert file paths to relative or absolute paths
```diff
-        <mesh filename="package://manipulator_h_description/meshes/link_1.stl" scale="0.001 0.001 0.001"/>
+        <mesh filename="../meshes/link_1.stl" scale="0.001 0.001 0.001"/>
```
You can then use rlCoachMdl to test the model (first command-line parameter is scene geometry, second is first robot's kinematics)
```shell
$ rlCoachMdl manipulator_h.urdf manipulator_h.urdf
```
[image]

santdiego, 2020-07-27:
Thank you very much for your reply. Is there any possibility to do the same with ros2?

rickertm, 2020-07-27:
For ROS2 you can use
```shell
$ ros2 run xacro xacro --inorder -o manipulator_h.urdf manipulator_h.xacro
```
The xacro package seems to have been added to Foxy Fitzroy on [2020-06-25](https://discourse.ros.org/t/new-packages-for-foxy-fitzroy-2020-06-25/15071/1).

santdiego, 2020-08-03:
> You can use `rl::mdl::UrdfFactory` and `rl::sg::UrdfFactory` for loading kinematics and geometry models from URDF files. If the URDF file uses ROS' xacro XML macro language, you first have to [...]

Hello, thank you very much for the responses. I followed the instructions in this answer in detail. I generated the robot model in urdf. However, in ubuntu 20.04 I get a "segmentation fault (core dumped)" error every time I try to start the precompiled executable "rlCoachMdl" with the * .urdf as argument. In windows it also doesn't start rlCoachMdl precompiled with this urdf.
When I try to recompile rlCoachMdl I get this same error in Windows and Linux:

`Running C:\Program Files\CMake\bin\cmake.exe "-GCodeBlocks - Ninja" C:/Users/i72GBSentey/Documents/GitHub/rl/demos/rlCoachMdl in C:\Users\i72GBSentey\Documents\GitHub\rl\demos\build-rlCoachMdl-Desktop_Qt_5_14_2_MSVC2015_64bit-Release.
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning at C:/Program Files/Robotics Library/0.7.0/MSVC/14.1/x64/lib/cmake/Coin-4.0.0a/CoinConfig.cmake:55 (message):
  Coin was generated with Visual Studio 15 2017 Win64 , not Ninja .
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  C:/Program Files/Robotics Library/0.7.0/MSVC/14.1/x64/lib/cmake/SoQt-1.6.0a/soqt-config.cmake:35 (find_dependency)
  CMakeLists.txt:2 (find_package)

-- Found unsuitable Qt version "5.14.2" from C:/Qt/5.14.2/msvc2015_64/bin/qmake.exe
**Qt QTOPENGL library not found.
Qt QTGUI library not found.
Qt QTNETWORK library not found.
Qt QTCORE library not found.**
CMake Error in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.18)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".

-- Configuring incomplete, errors occurred!
See also "C:/Users/i72GBSentey/Documents/GitHub/rl/demos/build-rlCoachMdl-Desktop_Qt_5_14_2_MSVC2015_64bit-Release/CMakeFiles/CMakeOutput.log".
CMake process exited with exit code 1.
Elapsed time: 00:00.`

rickertm, 2020-08-06:
The `rl::sg::UrdfFactory` was added after version 0.7, the precompiled executables of that version therefore do not yet support URDF files. On Ubuntu, you can however use the [daily PPA builds](https://code.launchpad.net/~roblib/+archive/ubuntu/daily) for a binary of the current master branch version.

When building RL from source, please configure CMake with the `CMakeLists.txt` in the project's root folder. The files in the demo folders require modules from other folders and therefore cannot be used directly. You can however limit the build to a specific target by specifying this at the command line:
```console
$ cmake --build <build_directory> --config Release --target rlCoachMdl
```