roboticslibrary/rl #28: 'geometry not supported' happened when loading scene

allenyh, 2020-07-31:
Hello, 
I build a project with VS 2019 running [rlRrtDemo](https://github.com/roboticslibrary/rl/blob/master/demos/rlRrtDemo/rlRrtDemo.cpp), 
the only change is that I replace the scene&kinematic loading mechanism from loading from terminal to loading from absolute path.

The modified loading part's code became like
`
...
std::string scenefile("C:\\Program Files\\Robotics Library\\0.7.0\\MSVC\\14.1\\x64\\share\\rl-0.7.0\\examples\\rlsg\\universal-robots-ur5_tables.xml");
...
if ("urdf" == scenefile.substr(scenefile.length() - 4, 4))
		{
			rl::sg::UrdfFactory factory;
			factory.load(scenefile, &scene);
		}
		else
		{
			rl::sg::XmlFactory factory;
			factory.load(scenefile, &scene);
		}

		std::string kinematicsfile("C:\\Program Files\\Robotics Library\\0.7.0\\MSVC\\14.1\\x64\\share\\rl-0.7.0\\examples\\rlmdl\\universal-robots-ur5.xml");
		std::shared_ptr<rl::mdl::Kinematic> kinematic;

		if ("urdf" == kinematicsfile.substr(kinematicsfile.length() - 4, 4))
		{
			rl::mdl::UrdfFactory factory;
			kinematic = std::dynamic_pointer_cast<rl::mdl::Kinematic>(factory.create(kinematicsfile));
		}
		else
		{
			rl::mdl::XmlFactory factory;
			kinematic = std::dynamic_pointer_cast<rl::mdl::Kinematic>(factory.create(kinematicsfile));
		}
...
`
and the running result said that 
_rl::sg::solid::Shape::create(SoVRMLShape* shape) - geometry not supported_

Is there anyone encounter the same error?

I build this project in Win10 with VS2019 and install RL using the windows installer.

allenyh, 2020-07-31:
BTW, The error only show up when I build the project under Release & RelWithDebInfo mode but not under Debug mode.
Is it due to the windows installer build RL under Debug mode?

rickertm, 2020-07-31:
The rlRrtDemo with the files `rlsg/universal-robots-ur5_tables.xml` and `rlmdl/universal-robots-ur5.xml` from rl-examples works fine on my installation, both with the current master branch and version 0.7 (with binaries from the 0.7 Windows installer). The latter however requires some modifications, as the code above for instance uses `rl::sg::XmlFactory`, which was not yet included in version 0.7. The Windows installer also only includes a release build.

I am only able to reproduce your error when mixing the 0.7 binaries from the installer with a different Coin3D version, where the type ids seem to be different. Please verify that you are compiling against the 0.7 version with the included headers and binaries and that you are not mixing versions.

allenyh, 2020-08-05:
@rickertm thanks for your reply
The reason that I can call XmlFactory is because I follow the tutorial [Build RL on Windows](https://www.roboticslibrary.org/tutorials/build-windows/).
It first use windows installer to install all the 3rd party libraries and also RL, and then use terminal to build and install this repo.
After several trials, it turns out that everything going right when compiling the project with "Debug" mode, while the error still coming out when compiling with other mode.

rickertm, 2020-08-06:
If I understand you correctly, you installed RL via the Windows installer (which includes third-party dependencies) and then also built the dependencies yourself as described in the [Building the Dependencies](https://www.roboticslibrary.org/tutorials/build-windows/#dependencies) tutorial?

Please make sure to only use one set of third-party binaries when building RL and running executables. In particular, do not install different versions in the same folder and make sure that your `PATH` variable is updated accordingly. As mentioned above, it looks like your are mixing Coin3D versions from the installer and your own build. As the debug files are not included in the Windows installer, this seems to use the correct version.