roboticslibrary/rl #29: ubuntu18.04 and  Qt5.14.1   cmake error

424358225, 2020-09-10:
Could NOT find RTAI (missing: RTAI_INCLUDE_DIRS) 
Could NOT find Xenomai (missing: Xenomai_INCLUDE_DIRS Xenomai_NATIVE_LIBRARY Xenomai_XENOMAI_LIBRARY) 

hello  gays,
It looks like ATIDAQ library not install.
What Can i do?

424358225, 2020-09-10:
https://github.com/roboticslibrary/atidaq 
i don,t konw how to build them.

rickertm, 2020-09-10:
RTAI and Xenomai are not related to the ATIDAQ library. These refer to optional components in rl::util for the respective real-time kernels.

The [ATIDAQ library](https://www.ati-ia.com/Products/ft/software/daq_software.aspx) is required for the `rl::hal::Ati` device class for force/torque sensors from ATI Industrial Automation. It is used to read calibration data from the files included with the sensor. To build the library, you can use the [CMake branch of the repository](https://github.com/roboticslibrary/atidaq/tree/cmake) which adds a CMakeLists.txt to the original sources:
```console
$ mkdir build
$ cd build
$ cmake ..
$ make
```

424358225, 2020-09-11:
Yes, I saw it.
[https://www.roboticslibrary.org/external-dependencies.svg](url)

sudo add-apt-repository ppa:roblib/ppa
sudo apt-get update

Mainly because I can't download them from the internet here.
Is there another way to download them?

424358225, 2020-09-11:
maybe china  firewall .
I try to use vpn,But the download still fails.

rickertm, 2020-09-11:
Apart from the RL itself, the [PPA on Launchpad](https://launchpad.net/~roblib/+archive/ubuntu/ppa) at the moment only includes packages for the optional collision engines [PQP](https://github.com/GammaUNC/PQP) and [SOLID](https://github.com/dtecta/solid3). The other packages listed in the tutorial should be available directly in the Ubuntu repositories.

Both PQP and SOLID now include CMake build scripts upstream, so you should be able to compile them similar to the ATIDAQ example above if you cannot access the PPA from your location.

closed 2020-09-23 by 424358225