roboticslibrary/rl #30: how to build scence for RL?

bennyfoo, 2020-09-17:
Hi there, How to build a simulation scene? VRML is not easy to build a scene.

rickertm, 2020-09-17:
You do not necessarily have to manually create a VRML file, there are a number of [3D modelers that support import and export of this format](https://www.web3d.org/x3d/content/examples/X3dResources.html#Export). Blender also used to include support for VRML export, but this addon was apparently removed with version 2.79.

If you have a more recent version of Blender, it still allows export to the closely related X3D format, the successor of VRML. This format can also be converted to VRML via [various tools and converters](https://www.web3d.org/x3d/content/examples/X3dResources.html#Conversions).

The simplest option for this seems to be a translation of the X3D XML format to VRML via the [XSLT Stylesheets](https://www.web3d.org/x3d/content/examples/X3dResources.html#Stylesheets) provided by the Web3D Consortium. The [current version](https://sourceforge.net/p/x3d/code/HEAD/tree/www.web3d.org/x3d/stylesheets/X3dToVrml97.xslt) requires an XSLT processor that supports version 2.0. An [older version](https://sourceforge.net/p/x3d/code/24039/tree/www.web3d.org/x3d/stylesheets/X3dToVrml97.xslt) however still uses version 1.1 and can be used with `xsltproc` included in RL's [libxslt](http://xmlsoft.org/libxslt/) dependency:
```console
$ xsltproc -o file.wrl X3dToVrml97.r24039.xslt file.x3d
```
These stylesheets are also included in the [X3D-Edit](https://savage.nps.edu/X3D-Edit/) editor and can be used to convert a file via the menu entry _X3D > Export from X3D > Export .x3d as VRML97 (.wrl)..._ if you prefer to use a GUI.

bennyfoo, 2020-09-18:
@   thanks for your reply

> You do not necessarily have to manually create a VRML file, there are a number of [3D modelers that support import and export of this format](https://www.web3d.org/x3d/content/examples/X3dResources.html#Export). Blender also used to include support for VRML export, but this addon was apparently removed with version 2.79.
> 
> If you have a more recent version of Blender, it still allows export to the closely related X3D format, the successor of VRML. This format can also be converted to VRML via [various tools and converters](https://www.web3d.org/x3d/content/examples/X3dResources.html#Conversions).
> 
> The simplest option for this seems to be a translation of the X3D XML format to VRML via the [XSLT Stylesheets](https://www.web3d.org/x3d/content/examples/X3dResources.html#Stylesheets) provided by the Web3D Consortium. The [current version](https://sourceforge.net/p/x3d/code/HEAD/tree/www.web3d.org/x3d/stylesheets/X3dToVrml97.xslt) requires an XSLT processor that supports version 2.0. An [older version](https://sourceforge.net/p/x3d/code/24039/tree/www.web3d.org/x3d/stylesheets/X3dToVrml97.xslt) however still uses version 1.1 and can be used with `xsltproc` included in RL's [libxslt](http://xmlsoft.org/libxslt/) dependency:
> 
> ```
> $ xsltproc -o file.wrl X3dToVrml97.r24039.xslt file.x3d
> ```
> 
> These stylesheets are also included in the [X3D-Edit](https://savage.nps.edu/X3D-Edit/) editor and can be used to convert a file via the menu entry _X3D > Export from X3D > Export .x3d as VRML97 (.wrl)..._ if you prefer to use a GUI.
> 
> The current master branch of RL also includes a `rl::sg::UrdfFactory` if you prefer to use this file format instead. It is able to parse URDF scene descriptions that use STL geometries.

closed 2020-09-18 by bennyfoo