roboticslibrary/rl #31: run the first applications

csufangyu, 2020-10-06:
hi when i use your first  applications on https://www.roboticslibrary.org/tutorials/first-steps-linux/,I get a error :+1: 
error: cannot dynamic_cast ‘rl::mdl::Factory::create(const string&)(std::__cxx11::basic_string<char>(((const char*)"/usr/share/rl-0.7.0/examples/rlmdl/unimation-puma560.xml"), std::allocator<char>()))’ (of type ‘class std::shared_ptr<rl::mdl::Model>’) to type ‘class rl::mdl::Kinematic*’ (source is not a pointer)
  rl::mdl::Kinematic* kinematics = dynamic_cast<rl::mdl::Kinematic*>(factory.create("/usr/share/rl-0.7.0/examples/rlmdl/unimation-puma560.xml"));

could you help me?thanks very much

csufangyu, 2020-10-06:
change :
`    std::shared_ptr<rl::mdl::Kinematic> kinematics = std::dynamic_pointer_cast<rl::mdl::Kinematic>(factory.create("/usr/share/rl-0.7.0/examples/rlmdl/unimation-puma560.xml"));`
is ok!

rickertm, 2020-10-17:
Thank you for pointing this out. I have updated the tutorials with code that should work for both the 0.7 and master branch.