roboticslibrary/rl #34: libxml2 build issues (MinGW)

githubuser0xFFFF, 2020-11-13:
Hi,

I try to build roboticslibrary with MinGW compiler. The first step was compiling the 3rdparty project. Compiling libxml2 caused a number of issues. I could work around two issues but now I'm stuck with a link problem. The following issues occured:

**Issue 1**

-- Build files have been written to: C:/CodingXP/roboticslibrary/3rdparty/build/mingw32/libxml2/libxml2-prefix/src/libxml2-build
[ 83%] Performing build step for 'libxml2'
Scanning dependencies of target LibXml2
[  1%] Building C object CMakeFiles/LibXml2.dir/buf.c.obj
[  2%] Building C object CMakeFiles/LibXml2.dir/c14n.c.obj
[  3%] Building C object CMakeFiles/LibXml2.dir/catalog.c.obj
In file included from C:/CodingXP/mingw730_32/i686-w64-mingw32/include/sys/stat.h:14:0,
                 from C:\CodingXP\roboticslibrary\3rdparty\build\mingw32\libxml2\libxml2-prefix\src\libxml2\catalog.c:23:
C:/CodingXP/mingw730_32/i686-w64-mingw32/include/io.h:280:34: error: macro "mkdir" requires 2 arguments, but only 1 given
   int __cdecl mkdir (const char *) __MINGW_ATTRIB_DEPRECATED_MSVC2005;
                                  ^
make[5]: *** [CMakeFiles\LibXml2.dir\build.make:91: CMakeFiles/LibXml2.dir/catalog.c.obj] Error 1
make[4]: *** [CMakeFiles\Makefile2:227: CMakeFiles/LibXml2.dir/all] Error 2
make[3]: *** [Makefile:151: all] Error 2
make[2]: *** [libxml2\CMakeFiles\libxml2.dir\build.make:117: libxml2/libxml2-prefix/src/libxml2-stamp/libxml2-build] Error 2
make[1]: *** [CMakeFiles\Makefile2:859: libxml2/CMakeFiles/libxml2.dir/all] Error 2
make: *** [Makefile:105: all] Error 2

**Solution**
Uncommenting line 301 in config.h solved this issue:

```c++
#if defined(_MSC_VER)
#define mkdir(p,m) _mkdir(p)
#if _MSC_VER < 1900
#define snprintf _snprintf
#endif
#if _MSC_VER < 1500
#define vsnprintf(b,c,f,a) _vsnprintf(b,c,f,a)
#endif
#elif defined(__MINGW32__)
//#define mkdir(p,m) _mkdir(p)
#endif
```

**Issue 2**

```
C:\CodingXP\mingw730_32\bin\windres.exe: C:\\CodingXP\\roboticslibrary\\3rdparty\\build\\mingw32\\libxml2\\libxml2-prefix\\src\\libxml2\\win32\\libxml2.rc:22: syntax error
make[5]: *** [CMakeFiles\LibXml2.dir\build.make:692: CMakeFiles/LibXml2.dir/win32/libxml2.rc.obj] Error 1
make[4]: *** [CMakeFiles\Makefile2:227: CMakeFiles/LibXml2.dir/all] Error 2
make[3]: *** [Makefile:151: all] Error 2
make[2]: *** [libxml2\CMakeFiles\libxml2.dir\build.make:117: libxml2/libxml2-prefix/src/libxml2-stamp/libxml2-build] Error 2
make[1]: *** [CMakeFiles\Makefile2:859: libxml2/CMakeFiles/libxml2.dir/all] Error 2
make: *** [Makefile:105: all] Error 2
```

**Solution**
Manually entering the version number into the version string for `ProductVersion` and `FileVersion` and removing the `LIBXML_DOTTED_VERSION` macro fixed the issue

```c++
BEGIN
  BLOCK "StringFileInfo"
  BEGIN
    BLOCK "04090000"  /* Lang = US English, Charset = ASCII */
    BEGIN
      VALUE "FileDescription",	"libxml2 library\0"
      VALUE "FileVersion",	"2.9.9\0"
      VALUE "InternalName",	"libxml2.dll\0"
      VALUE "LegalCopyright",	"Copyright (C) Daniel Veillard\0"
      VALUE "LegalTrademarks",	"\0"
      VALUE "OriginalFilename",	"libxml2.dll\0"
      VALUE "ProductName",	"libxml2\0"
      VALUE "ProductVersion",	"2.9.9\0"
      VALUE "Comments",		"For more information visit http://xmlsoft.org/\0"
```

**Issue 3**

```
[ 52%] Linking C shared library libxml2.dll
CMakeFiles\LibXml2.dir/objects.a(xmlcatalog.c.obj):xmlcatalog.c:(.text+0x446): undefined reference to `_imp__xmlFree'
CMakeFiles\LibXml2.dir/objects.a(xmlcatalog.c.obj):xmlcatalog.c:(.text+0x4c8): undefined reference to `_imp__xmlFree'
CMakeFiles\LibXml2.dir/objects.a(xmlcatalog.c.obj):xmlcatalog.c:(.text+0x6ea): undefined reference to `_imp__xmlFree'
CMakeFiles\LibXml2.dir/objects.a(xmlcatalog.c.obj):xmlcatalog.c:(.text+0x176f): undefined reference to `_imp__xmlFree'
CMakeFiles\LibXml2.dir/objects.a(xmlcatalog.c.obj):xmlcatalog.c:(.text+0x1838): undefined reference to `_imp__xmlFree'
CMakeFiles\LibXml2.dir/objects.a(xmlcatalog.c.obj):xmlcatalog.c:(.text+0x1856): more undefined references to `_imp__xmlFree' follow
collect2.exe: error: ld returned 1 exit status
make[5]: *** [CMakeFiles\LibXml2.dir\build.make:817: libxml2.dll] Error 1
make[4]: *** [CMakeFiles\Makefile2:227: CMakeFiles/LibXml2.dir/all] Error 2
make[3]: *** [Makefile:151: all] Error 2
make[2]: *** [libxml2\CMakeFiles\libxml2.dir\build.make:117: libxml2/libxml2-prefix/src/libxml2-stamp/libxml2-build] Error 2
make[1]: *** [CMakeFiles\Makefile2:859: libxml2/CMakeFiles/libxml2.dir/all] Error 2
make: *** [Makefile:105: all] Error 2
```

I tried the proposed solution here: 

http://gnome-lib.996298.n3.nabble.com/Problem-with-MSYS-and-MINGW-undefined-reference-to-imp-xmlFree-td6546.html

But this did not fix the issue. Does anyone know how to fix this issue?

rickertm, 2020-11-13:
These issues should be fixed in the current libxml2 branch [upstream](https://gitlab.gnome.org/GNOME/libxml2/), which now includes CMake files. However, there is currently no new release version with these patches. I've updated the third-party CMake files to use a commit that includes these patches. Please check if this works for you.

MinGW should also include packages for most of RL's dependencies, including libxml2. The only ones missing are the optional collision detection engines FCL, ODE, PQP, SOLID3.

rickertm, 2020-12-02:
[MSYS2](https://www.msys2.org/) now also includes MinGW-w64 packages for ODE, PQP and SOLID3. I've added a corresponding [tutorial for building RL](https://www.roboticslibrary.org/tutorials/build-mingw-w64/) on this platform to the website.

githubuser0xFFFF, 2020-12-02:
Thank you for your support. I will try to rebuild the library.

closed 2020-12-02 by githubuser0xFFFF