roboticslibrary/rl #35: DOF?

eroswade, 2020-12-07:
it seems rl::kin support free DOF
but rl::mdl support 6 DOF only?

rickertm, 2020-12-09:
Both rl::kin and rl::mdl support tree-like kinematic structures with multiple joints. rl::kin supports revolute and prismatic joints in classical DH notation, while rl::mdl supports revolute, prismatic, cylindrical, helical, spherical, and 6-DOF joints.

The [rl-examples](https://github.com/roboticslibrary/rl-examples) repository includes various example models for rl::mdl, including 7-DOF manipulators and a dual-arm manipulator plus hands with a total of 41 DOF.

eroswade, 2020-12-24:
1.  is kin and mdl accept less than 6 DOF,.  3 DOF for example?
i find that in mdl JacobianInverseKinematics line 104
::rl::math::VectorBlock dxi = dx.segment(6 * this->goals[i].second, 6);  // is that  means not support less than 6 DOF?
2. is there any example that can init kinematics using DH?

rickertm, 2020-12-26:
RL uses a 3D representation for transformations (translation and rotation). A delta vector in the kinematics calculations is therefore represented as a 6D vector (3 values for translation and 3 values for rotation). It also supports tree-like kinematic structures, therefore a kinematic model can have more than one end-effector. Line 104 mentioned above refers to the 6D delta vector of end-effector _i_ of a model. The controllable degrees of freedom of an end-effector depend on the joints and structure of the model.

rl::kin uses the classical DH notation by Paul and only supports revolute and prismatic joints. A number of example models are included in the [rl-examples](https://github.com/roboticslibrary/rl-examples) repository. rl::mdl uses Spatial Vector Algebra to support multiple joint types with a single notation and is not limited to changes in 4 dimensions as DH. You can still define a model based on DH parameters by combining multiple basic _fixed_, _revolute_, _frame_, _body_ elements, e.g., DH (Paul) with _d_, _theta_, _a_, _alpha_ as in our [tutorial](https://www.roboticslibrary.org/tutorials/create-a-robot-model/). However, you can also simplify this and define axes that rotate around _x_ or _y_ instead of _z_. Multiple end-effectors are supported as well, see [mekabot-torso-rightarm-leftarm.xml](https://github.com/roboticslibrary/rl-examples/blob/master/rlmdl/mekabot-torso-rightarm-leftarm.xml) or [mekabot-torso-rightarm-righthand-leftarm-lefthand.xml](https://github.com/roboticslibrary/rl-examples/blob/master/rlmdl/mekabot-torso-rightarm-righthand-leftarm-lefthand.xml).