roboticslibrary/rl #37: Load a scene from urdf crashed

liufang-robot, 2020-12-15:
Hello,I'm using latest RL from source building.
When I try to load a scene from a urdf file as follow:
```
std::shared_ptr<rl::sg::Factory> factory = std::make_shared<rl::sg::UrdfFactory>();
std::shared_ptr<rl::sg::Scene> scene = std::make_shared<rl::sg::bullet::Scene>();
factory->load("my_scene.urdf", scene.get());
```
it crashed when calling the `load` function and showing:
```
SoBase.cpp:199: SoBase::SoBase(): Assertion `(SoBase::classTypeId != SoType::badType()) 
&& "An SoBase-derived class was attempted instantiated *before* Coin initialization. 
(Have you perhaps placed an SoBase-derived instance (e.g. a scene graph node) in non-heap memory?) 
See SoBase class documentation for more info."' failed.
```

I traced and found it crashed at `::SoVRMLShape* vrmlShape = new ::SoVRMLShape();` in file `UrdfFactory.cpp`.
It seems the problem is one must call some coin init function before creating any vrml things. Am I correct?

Do I miss something or What should I do?
Thanks.

rickertm, 2020-12-16:
Thank you for reporting this. You are correct, the Coin3D libraries first need to be initialized before creating SoBase-derived classes. For `rl::sg::XmlFactory` this is handled via `SoInput`, while the GUI demos call `SoQt::init()` and `SoDB::init()` explicitly. `rl::sg::UrdfFactory` also requires calling `SoNodeKit::init()` due to `SoSTLFileKit`.

This should be fixed by e069d223c2ca8d3c87d9b5880007f3c13196f3f7, that adds explicit calls to these functions to the scene graph factories. Alternatively, you can also add the following two lines before creating any Coin3D objects:
```cpp
SoDB::init();
SoNodeKit::init();
```

liufang-robot, 2020-12-16:
This works. Now the scene is loaded.

closed 2020-12-16 by liufang-robot