roboticslibrary/rl #38: Self collision problem

liufang-robot, 2020-12-17:
Hello, I have a problem.
I load the scene (official puma560) case as follow:

```
# remove the boxes model in unimation-puma560_boxes.xml
std::shared_ptr<rl::sg::Factory> factory = std::make_shared<rl::sg::XmlFactory>();
std::shared_ptr<rl::sg::Scene> scene = std::make_shared<rl::sg::bullet::Scene>();
factory->load("unimation-puma560_boxes.xml", scene.get());
rl::sg::SimpleScene* simpleScene = dynamic_cast<rl::sg::SimpleScene*>(scene.get());
std::cout<<"is colliding "<<simpleScene->isColliding()<<"\n";
```
**I removed the `boxes` model in `unimation-puma560_boxes.xml`.**
Then check if the scene is in collision.
the result shows that the scene is in collision.
I believe it's because some conjunctive link models in puma560 are in collision.

I know that in mdl xml file, there is an `ignore` tag in body, which can be read from kinematic api.
But the `SimpleScene` don't use the kinematic information directly.

So the problem is what is the best way to check the scene collision considering the conjunctive link ignore information?

Shortly speaking, I want to quicked get the result that the above case is not really in collision.

Thanks.

rickertm, 2020-12-18:
The self collision properties are currently part of the kinematic descriptions, therefore the scene graph abstraction does not have access to this information. Kinematic and geometric models are combined in `rl::plan::SimpleModel`, which provides a function `bool isColliding(const ::rl::math::Vector& q)` that you can use:
```cpp
rl::plan::SimpleModel simpleModel;
simpleModel.kin = kinematicModel; // for rl::kin::Kinematics*
simpleModel.mdl = kinematicModel; // for rl::mdl::Kinematic*
simpleModel.scene = simpleScene;
simpleModel.model = simpleScene->getModel(0); // id of kinematic model in scene
bool isColliding = simpleModel.isColliding(q);
```

liufang-robot, 2020-12-20:
thanks for your reply.
I think I get it, just use the `SimpleModel` in `plan` to do the collision check.It will take into account the ignore ref information of kinematics(I have checked the source code).