roboticslibrary/rl #39: Up-to-date build instructions

githubuser0xFFFF, 2021-02-05:
Hi,

are there any up-to-date build instructions available to build the library from source? I just noticed that all CMakeLists.txt files are missing from the 3rdparty folder. Does this mean, that rl expects that all 3rdparty libraries are alredy installed?

The last time I tried to build the library, I started with the 3rdparty folder like written in the build instructions. Is this still the right way?

Thank you,

Uwe

rickertm, 2021-02-05:
I've moved the 3rdparty folder to a separate repository starting with 593ec79d0c6e86d855b6b4a941f87055d1c9dbad:
https://github.com/roboticslibrary/rl-3rdparty

This repository now has a separate CI build to generate binaries with the most recent versions of the dependencies to hopefully make this step easier for users and to reuse them in RL's CI build for MSVC. You can download a 7Z archive and an MSI installer here:
https://github.com/roboticslibrary/rl-3rdparty/releases/tag/latest

githubuser0xFFFF, 2021-02-06:
Thank you for the quick reply - then I can start the adventure now of building the library with MinGW (without MSYS).

closed 2021-02-06 by githubuser0xFFFF

githubuser0xFFFF, 2021-02-06:
@rickertm Hi Markus, thanks to your help I managed to build the 3rdparty libraries and the rl library and I could successfully run 

```
rlCoachMdl unimation-puma560_boxes.xml unimation-puma560.xml
```
I would like to give you some feedback about the build process and some issues I had.

### libxml2 issues
Build was a lot smoother than the last time I tried the build (see issue #34). I had only one single error:

```
[ 76%] Building RC object CMakeFiles/liblzma.dir/src/liblzma/liblzma_w32res.rc.obj
C:\CodingXP\mingw730_64\bin\windres.exe: C:\\CodingXP\\roboticslibrary_3rdparty\\build\\mingw64\\xz\\xz-prefix\\src\\xz\\src\\common/common_w32res.rc:44: syntax error
make[5]: *** [CMakeFiles\liblzma.dir\build.make:1117: CMakeFiles/liblzma.dir/src/liblzma/liblzma_w32res.rc.obj] Error 1
make[5]: *** Waiting for unfinished jobs....
make[4]: *** [CMakeFiles\Makefile2:98: CMakeFiles/liblzma.dir/all] Error 2
make[3]: *** [Makefile:148: all] Error 2
make[2]: *** [xz\CMakeFiles\xz.dir\build.make:134: xz/xz-prefix/src/xz-stamp/xz-build] Error 2
make[1]: *** [CMakeFiles\Makefile2:766: xz/CMakeFiles/xz.dir/all] Error 2
make: *** [Makefile:124: all] Error 2
```

**Solution**

I fixed this by editing the file `liblzma_w32res.rc`. I changed the line 44 - from `MY_PRODUCT` into `"MY_PRODUCT"`

```
BEGIN
    BLOCK "StringFileInfo"
    BEGIN
        BLOCK "040904b0"
        BEGIN
            VALUE "CompanyName", MY_COMPANY
            VALUE "FileDescription", MY_DESC
            VALUE "FileVersion", LZMA_VERSION_STRING
            VALUE "InternalName", MY_NAME
            VALUE "OriginalFilename", MY_FILENAME
            VALUE "ProductName", "MY_PRODUCT"
            VALUE "ProductVersion", LZMA_VERSION_STRING
        END
    END
```

### coin issues

Here was the problem, that I already have the boost libraries in my MinGW instalation so I don`t wanted to build boost. Coind depends on boost and if boost is not build in the 3rdparty project, then coin has problems to find boost and to build properly. In the 3rdparty project there is a boost dependency of coin that leads to a build of boost although boost is already present.

**Solution**

- remove `boost` from the `DEPENDS` statement in `roboticslibrary/3rdparty/coin3d/coin/CMakeLists.txt`
- remove `add_subdirectory(boost)` from the main `CMakeLists.txt` file (`roboticslibrary/3rdparty/CMakeLists.txt`)

Because boost dependency is now removed from the 3rdparty project the coin library does not find boost anymore.

**Solution**

Changed coin `CMakeLists.txt` file to disable `find_package(Boost)` and set hardcoded boost variables:

```cmake
# As not all CMake FindXXX modules and config packages provide target support,
# we should explicitly check for them on.
#find_package(Boost REQUIRED) # almost any Boost will do

set(Boost_INCLUDE_DIRS "C:/CodingXP/mingw730_32/usr/include")
set(Boost_LIBRARY_DIRS "C:/CodingXP/mingw730_32/usr/lib")

#Boost::boost - Target for header only dependencies (Boost include directory)
if(NOT TARGET Boost::boost)
  list(APPEND COIN_TARGET_INCLUDE_DIRECTORIES ${Boost_INCLUDE_DIRS})
else()
  list(APPEND COIN_TARGET_LINK_LIBRARIES Boost::boost)
endif()
```

### RL issues

When building the RL library I had the following errors when building the tests and demos (exe files):

```
Scanning dependencies of target rlNearestNeighborsTest
[ 32%] Building CXX object tests/rlNearestNeighborsTest/CMakeFiles/rlNearestNeighborsTest.dir/rlNearestNeighborsTest.cpp.obj
In file included from C:/CodingXP/roboticslibrary/build/mingw64/install/include/fcl/math/bv/kDOP.h:177:0,
                 from C:/CodingXP/roboticslibrary/build/mingw64/install/include/fcl/math/bv/utility-inl.h:46,
                 from C:/CodingXP/roboticslibrary/build/mingw64/install/include/fcl/math/bv/utility.h:61,
                 from C:/CodingXP/roboticslibrary/build/mingw64/install/include/fcl/broadphase/broadphase_dynamic_AABB_tree.h:44,
                 from C:\CodingXP\roboticslibrary\src\rl\sg\fcl\Body.h:35,
                 from C:\CodingXP\roboticslibrary\src\rl\sg\fcl\Body.cpp:27:
C:/CodingXP/roboticslibrary/build/mingw64/install/include/fcl/math/bv/kDOP-inl.h:83:1: error: function 'fcl::KDOP<S_, N>::KDOP()' definition is marked dllimport
 KDOP<S, N>::KDOP()
 ^~~~~~~~~~
 ```
This is a problem I have also seen building other libraries with CMake and MinGW but I don't know how to fix this properly and why this happens. Normall MinGW does not need `dllimport` and `dllexport` statements because it can link directly to DLLs. The dllexport and dlimport statements are only required for MSVC but CMake also generates them for MinGW. I could only fix this by manually editing all `export.h` files such as

```
fcl/export.h
sg/export.h
plan/export.h
mdl/export.h
kin/export.h
hal/export.h
```
I manually edited the files to set the export defines to an empty value to disable `dllimport` and `dllexport` statement. The following code snippet shows the file `fcl/export.h` I added the empty `#define FCL_EXPORT` statement in line 4. I did this for all the export.h files mentioned above. I have no clue, how to fix this properly. I think I have to write into the CMake support list to learn how to fix this.

```c++
#ifndef FCL_EXPORT_H
#define FCL_EXPORT_H

#  define FCL_EXPORT

#ifdef FCL_STATIC_DEFINE
#  define FCL_EXPORT
#  define FCL_NO_EXPORT
#else
#  ifndef FCL_EXPORT
#    ifdef fcl_EXPORTS
        /* We are building this library */
#      define FCL_EXPORT __declspec(dllexport)
#    else
        /* We are using this library */
#      define FCL_EXPORT __declspec(dllimport)
#    endif
#  endif
```

After these changes, I succeeded building the RL library and could run the `rlMdlCoach` application.

rickertm, 2021-02-06:
Thank you very much for the detailed feedback on your experience when building this on MinGW.

> I fixed this by editing the file liblzma_w32res.rc. I changed the line 44 - from MY_PRODUCT into "MY_PRODUCT"

This file is included in [xz/liblzma](https://tukaani.org/xz/). Ideally this issue should be reported upstream. The project uses a mailing list for communication.

> Here was the problem, that I already have the boost libraries in my MinGW installation so I don`t wanted to build boost. Coin depends on boost and if boost is not build in the 3rdparty project, then coin has problems to find boost and to build properly. In the 3rdparty project there is a boost dependency of coin that leads to a build of boost although boost is already present.

The repository currently tries to setup all necessary dependencies (apart from Qt) without looking at existing installations. Disabling individual packages in CMakeLists.txt should be ok.

> Because boost dependency is now removed from the 3rdparty project the coin library does not find boost anymore.

Have you tried setting `CMAKE_PREFIX_PATH` (or `BOOST_INCLUDEDIR`, `BOOSTROOT`) to point to your local Boost installation? CMake 3.19.4 does not yet know about Boost 1.75, this was only added a few days ago, however it should still be able to find the headers.

> C:/CodingXP/roboticslibrary/build/mingw64/install/include/fcl/math/bv/kDOP-inl.h:83:1: error: function 'fcl::KDOP<S_, N>::KDOP()' definition is marked dllimport
> This is a problem I have also seen building other libraries with CMake and MinGW but I don't know how to fix this properly and why this happens. Normally MinGW does not need dllimport and dllexport statements because it can link directly to DLLs. The dllexport and dlimport statements are only required for MSVC but CMake also generates them for MinGW. I could only fix this by manually editing all export.h files such as

Does this only show up for FCL or also in other locations? There is an open issue for FCL when building it as a shared library due to adding dllexport/dllimport to templates that has not been fixed yet.

RL uses the CMake module [`GenerateExportHeader`](https://cmake.org/cmake/help/latest/module/GenerateExportHeader.html) for generating the export headers. So far I have not seen any issues on Linux/macOS/MSVC/MSYS2. If you think this is a generic CMake issue on MinGW it might be a good idea to let them know.

githubuser0xFFFF, 2021-02-18:
@rickertm Hi Markus,

thank you for your answer. When using the 3rdparty libraries and the roboticslibrary that has been build in release mode in an application build in debug mode, the  the following error occurs:

```
QObject::installEventFilter(): Cannot filter events for objects in a different thread.
QObject::setParent: Cannot set parent, new parent is in a different thread
ASSERT failure in QCoreApplication::sendEvent: "Cannot send events to objects owned by a different thread. Current thread 0x0x4beaa6d0. Receiver 'QtGLArea' (of type 'SoQtGLArea') was created in thread 0x0xb565e8e0", file kernel\qcoreapplication.cpp, line 577 (kernel\qcoreapplication.cpp:577, (null))
```

To solve this, I needed to rebuild the 3rdparty libraries and the roboticslibrary in debug mode. This caused the following additional build issues:

### Link Issue: Too many sections

Building the fcl library and the roboticslibrary caused the following link issues:

```
[ 97%] Linking CXX shared library ..\..\..\bin\librlsg.dll
[ 97%] Built target sg
C:/CodingXP/mingw730_64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/../../../../x86_64-w64-mingw32/bin/as.exe: CMakeFiles\kin.dir\Kinematics.cpp.obj: too many sections (35500)
C:\Users\UWEKIN~1\AppData\Local\Temp\ccg2gkUf.s: Assembler messages:
C:\Users\UWEKIN~1\AppData\Local\Temp\ccg2gkUf.s: Fatal error: can't write 171 bytes to section .text of CMakeFiles\kin.dir\Kinematics.cpp.obj: 'File too big'
C:/CodingXP/mingw730_64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/../../../../x86_64-w64-mingw32/bin/as.exe: CMakeFiles\kin.dir\Kinematics.cpp.obj: too many sections (35500)
```
**Solution**

I could solve the issue by adding `-Wa,-mbig-obj` to the `CMAKE_CXX_FLAGS`.

### Missing debug postfix

When building the roboticslibrary I noticed that the debug postfix for the libraries was missing. I fixed this by setting `CMAKE_DEBUG_POSTFIX` to `d`.

Best regards, Uwe

rickertm, 2021-02-23:
> ASSERT failure in QCoreApplication::sendEvent: "Cannot send events to objects owned by a different thread. Current thread 0x0x4beaa6d0. Receiver 'QtGLArea' (of type 'SoQtGLArea') was created in [...]

Does this only show up when linking a debug build against a release version? Is this with one of the demo applications? The `SoQtGLArea` class is part of SoQt. When updating a model from a different thread, Qt typically requires using signals and slots.

> I could solve the issue by adding -Wa,-mbig-obj to the CMAKE_CXX_FLAGS.

As you pointed out, most likely due to FCL. I will add this to the FAQ section of the website.

> When building the roboticslibrary I noticed that the debug postfix for the libraries was missing. I fixed this by setting CMAKE_DEBUG_POSTFIX to d.

Earlier versions added a `d` postfix for all compilers on `WIN32`. I limited this to `MSVC` in 071a84b8515a4405ca2955d2417644b9090280ec, which also sets an `s`  postfix for static builds.