roboticslibrary/rl #40: Depth Collid Error~

424358225, 2021-02-23:
`if (rl::sg::SimpleScene* simpleScene = dynamic_cast<rl::sg::SimpleScene*>(control_scene.get()))---> true
if (rl::sg::DepthScene* depthScene = dynamic_cast<rl::sg::DepthScene*>(control_scene.get()))--->false
if (rl::sg::DistanceScene* distanceScene = dynamic_cast<rl::sg::DistanceScene*>(control_scene.get()))--->true`

Only DepthScene Not Success!

rickertm, 2021-02-23:
What type of collision engine are you using as `control_scene`? Please make sure the engine you are using supports depth queries:
http://doc.roboticslibrary.org/0.7.0/d4/deb/classrl_1_1sg_1_1_depth_scene.html

424358225, 2021-02-24:
> What type of collision engine are you using as `control_scene`? Please make sure the engine you are using supports depth queries:
> http://doc.roboticslibrary.org/0.7.0/d4/deb/classrl_1_1sg_1_1_depth_scene.html

control_scene = std::make_shared<rl::sg::solid::Scene>();

424358225, 2021-02-24:
> What type of collision engine are you using as `control_scene`? Please make sure the engine you are using supports depth queries:
> http://doc.roboticslibrary.org/0.7.0/d4/deb/classrl_1_1sg_1_1_depth_scene.html

once again,Application crash

> qhull precision error: initial facet 1 is coplanar with the interior point
ERRONEOUS FACET:
- f1
    - flags: bottom simplicial flipped
    - normal:        -0       -1       -0
    - offset:     -0.734
    - vertices: p0 (v2) p3 (v1) p2 (v0)
    - neighboring facets:  f2 f3 f4

While executing:  | qhull Qts i Tv
Options selected for Qhull 2002.1 2002/8/20:
  Qtriangulate  Qsearch-initial-simplex  incidence  Tverify  _pre-merge
  _zero-centrum  _max-width 0.37  Error-roundoff 1.4e-15  _one-merge 9.7e-15
  _near-inside 4.8e-14  Visible-distance 2.8e-15  U-coplanar-distance 2.8e-15
  Width-outside 5.5e-15  _wide-facet 1.7e-14

precision problems (corrected unless 'Q0' or an error)
      2 flipped facets

The input to qhull appears to be less than 3 dimensional, or a
computation has overflowed.

Qhull could not construct a clearly convex simplex from points:
- p1 (v3):  0.63 -0.73  0.16
- p0 (v2):  0.63 -0.73  0.16
- p3 (v1):  0.99 -0.73 -0.16
- p2 (v0):  0.63 -0.73 -0.16

The center point is coplanar with a facet, or a vertex is coplanar
with a neighboring facet.  The maximum round off error for
computing distances is 1.4e-15.  The center point, facets and distances
to the center point are as follows:

center point   0.7202   -0.734 0.003201

facet p0 p3 p2 distance=    0
facet p1 p3 p2 distance=    0
facet p1 p0 p2 distance=    0
facet p1 p0 p3 distance=    0

These points either have a maximum or minimum x-coordinate, or
they maximize the determinant for k coordinates.  Trial points
are first selected from points that maximize a coordinate.

The min and max coordinates for each dimension are:
  0:    0.6252    0.9948  difference= 0.3697
  1:    -0.734  -2.225e-308  difference= 0.734
  2:   -0.1627    0.1648  difference= 0.3275

If the input should be full dimensional, you have several options that
may determine an initial simplex:
  - use 'QJ'  to joggle the input and make it full dimensional
  - use 'QbB' to scale the points to the unit cube
  - use 'QR0' to randomly rotate the input for different maximum points
  - use 'Qs'  to search all points for the initial simplex
  - use 'En'  to specify a maximum roundoff error less than 1.4e-15.
  - trace execution with 'T3' to see the determinant for each point.

If the input is lower dimensional:
  - use 'QJ' to joggle the input and make it full dimensional
  - use 'Qbk:0Bk:0' to delete coordinate k from the input.  You should
    pick the coordinate with the least range.  The hull will have the
    correct topology.
  - determine the flat containing the points, rotate the points
    into a coordinate plane, and delete the other coordinates.
  - add one or more points to make the input full dimensional.
munmap_chunk(): invalid pointer

rickertm, 2021-02-24:
The following example works fine on my system, both with current master and on branch 0.7:
```cpp
#include <iostream>
#include <memory>
#include <rl/sg/solid/Scene.h>

int
main(int argc, char** argv)
{
	std::shared_ptr<rl::sg::Scene> control_scene = std::make_shared<rl::sg::solid::Scene>();;
	
	if (rl::sg::SimpleScene* simpleScene = dynamic_cast<rl::sg::SimpleScene*>(control_scene.get()))
	{
		std::cout << "SimpleScene" << std::endl;
	}
	if (rl::sg::DepthScene* depthScene = dynamic_cast<rl::sg::DepthScene*>(control_scene.get()))
	{
		std::cout << "DepthScene" << std::endl;
	}
	if (rl::sg::DistanceScene* distanceScene = dynamic_cast<rl::sg::DistanceScene*>(control_scene.get()))
	{
		std::cout << "DistanceScene" << std::endl;
	}
	
	return EXIT_SUCCESS;
}
```
Output:
```
SimpleScene
DepthScene
DistanceScene
```
Please provide more details on your setup, compiler, plattform, dependencies etc. and a full (minimal) example to reproduce the issue.

> The input to qhull appears to be less than 3 dimensional, or a computation has overflowed.
> Qhull could not construct a clearly convex simplex from points:
>    p1 (v3): 0.63 -0.73 0.16
>    p0 (v2): 0.63 -0.73 0.16
>    p3 (v1): 0.99 -0.73 -0.16
>    p2 (v0): 0.63 -0.73 -0.16

Qhull is used by SOLID to construct a convex hull of a shape by default. In this case, two of the four points are identical and the shape is a flat polygon. Qhull cannot construct a convex hull for this shape. Please correct your geometry model or mark the shape as non-convex via the [`convex`](https://www.web3d.org/documents/specifications/14772/V2.0/part1/concepts.html#4.6.3.4) field of the geometry, SOLID will then create a `DT_NewComplexShape` shape instead.
```
      geometry IndexedFaceSet {
        convex FALSE
```

424358225, 2021-03-05:
> 以下示例适用于我的系统，无论是当前主系统还是分支 0.7：
> 
> ```c++
> #include <iostream>
> #include <memory>
> #include <rl/sg/solid/Scene.h>
> 
> int
> main(int argc, char** argv)
> {
> 	std::shared_ptr<rl::sg::Scene> control_scene = std::make_shared<rl::sg::solid::Scene>();;
> 	
> 	if (rl::sg::SimpleScene* simpleScene = dynamic_cast<rl::sg::SimpleScene*>(control_scene.get()))
> 	{
> 		std::cout << "SimpleScene" << std::endl;
> 	}
> 	if (rl::sg::DepthScene* depthScene = dynamic_cast<rl::sg::DepthScene*>(control_scene.get()))
> 	{
> 		std::cout << "DepthScene" << std::endl;
> 	}
> 	if (rl::sg::DistanceScene* distanceScene = dynamic_cast<rl::sg::DistanceScene*>(control_scene.get()))
> 	{
> 		std::cout << "DistanceScene" << std::endl;
> 	}
> 	
> 	return EXIT_SUCCESS;
> }
> ```
> 
> 输出：
> 
> ```
> SimpleScene
> DepthScene
> DistanceScene
> ```
> 
> 请提供有关您的设置、编译器、拼贴表、依赖项等的更多详细信息，以及复制问题的完整（最小）示例。
> 
> > 对 qhull 的输入似乎小于 3 维，或者计算已溢出。 Qhull 无法从点构建一个清晰的凸起简单：  p1 （v3）： 0.63 -0.73 0.16  p0 （v2）： 0.63 -0.73 0.16  p3 （v1）： 0.99  -0.73 -0.16 p2 （v0）： 0.63 -0.73 -0.16
> 
> 默认情况下，固体用于构建形状的凸体。在这种情况下，四个点中的两个相同，形状为平面多边形。Qhull 不能为这种形状建造凸体。请更正您的几何模型或通过几何[`的凸`](https://www.web3d.org/documents/specifications/14772/V2.0/part1/concepts.html#4.6.3.4)场将形状标记为非凸，然后SOLID将创建一个形状。`DT_NewComplexShape`
> 
> ```
>       geometry IndexedFaceSet {
>         convex FALSE
> ```

thanks a lot

closed 2021-03-05 by 424358225