roboticslibrary/rl #42: which link is the tool?

marcoojer, 2021-05-19:
Hello, i have a doubt, when doing a direct or inverse kinematics, how do you know which is the end_effector link. I saw that in the rl::mdl::Model there are a _leave and _tool member variables and when doing kinematics->getOperationalPosition(0), you do the calculus for the first _tool;. But I don't know how is the mapping between the index and the link name. How it is done?

rickertm, 2021-05-20:
A kinematic model is represented as a tree with vertices and edges. Vertices represent intermediate frames or bodies/links. End effectors are the leaves of the tree. The tree is parsed using depth-first search and each leaf is added to the end of the `leaves` vector in the model.

In order to specify goal transformations for multiple end effectors in the inverse kinematics calculation, you can call the `rl::mdl::InverseKinematics::addGoal()` function for each end effector transform matrix and index before starting the calculation with `rl::mdl::InverseKinematics::solve()`.

marcoojer, 2021-05-21:
Thank you very much  for answering, but how do I know the which link corresponds to each end_effector index, for example, if I have in my .urdf two "leave nodes" ```tool_0``` and ```tool_1```, how do i know which tool index correspond to each one in the ```rl::mdl::Kinematics```?

rickertm, 2021-05-21:
The function `rl::mdl::Model::getOperationalFrame(size_t& i)` gives you access to the `rl::mdl::Frame` instead of just its transformation matrix. This class has a `getName()` function that should help you in identifying the correct leaf.

marcoojer, 2021-05-24:
Thank you!! that solved my doubt!!

closed 2021-05-24 by marcoojer