roboticslibrary/rl #44: How do I make a simple simula

alexdupond, 2021-05-26:
Hi. I just started to look at this lib and went through the example code for setting up an simulation and also for the robot path planning. Now I want to combine them to replay the path on the in the simulation and I'm not sure what is the procedure for consciously updating the joint position of the robot in a scene. (I'm also very new to QT) 

All the tests I have done is on the puma560 robot and as I understand it I use the screen xml file for describing the screen and a model xml file for describing kinematics of the robot. I have made got the planner to work and the simulator to show the robot independently. So my question is, how do I I update the scene and the view so that I can simulate the robot move? Maybe some steps, some reading material that describes this or some small code snippet samples. 

Hope someone can point me in the right direction.

rickertm, 2021-05-27:
In order to visualize the robot at each configuration of your solution path, you will have to calculate the forward position kinematics at each point and synchronize the kinematic and geometric models for the robot. This is generally done by getting and setting each robot body's frame. In `rl::plan`, these two models are combined and updated in `rl::plan::Model`, which is also used in the `rlPlanDemo` application.
https://github.com/roboticslibrary/rl/blob/3c522c3abc2be819bdb6512cd5805f0a34d851d2/src/rl/plan/Model.cpp#L536-L544

Some example code on how to animate this is also available in `rlPlanDemo`
https://github.com/roboticslibrary/rl/blob/3c522c3abc2be819bdb6512cd5805f0a34d851d2/demos/rlPlanDemo/Thread.cpp#L389-L412

In order to animate this in Qt while still being able to interact with your GUI, you typically have to use a separate thread. As the widget with the 3D model is part of [Qt's main thread](https://doc.qt.io/qt-5/threads-qobject.html), you will have to use [Qt signals and slots](https://doc.qt.io/qt-5/signalsandslots.html) to communicate between them. This is also used in `rlPlanDemo`.

alexdupond, 2021-05-28:
Thanks for the fast reply and your time. I will try this and go through the rlPlanDemo again. I looks at it once, but got a little lost in all of the different drawview settings. I will go through it again in more details.

Thanks again :)