roboticslibrary/rl #45: Add Inverse Kinematics solver with multiple solutions

johnny9938, 2021-06-29:
Can you please add a new function to solve the inverse kinematics of a given robot, and get multiple possible solutions (usually 8 solutions)?

M1sail, 2022-06-28:
您好！
 I have encountered a similar problem. Since there are multiple inverse kinematics results for the same location in space, I am getting a different robot configuration each time after performing the inverse kinematics solution for the same section of trajectory.
Like this.
[image]
and
[image]

How do I set it up to get the desired robot configuration when using JacobianInverseKinematics?

Looking forward to your answer~
：）

rickertm, 2022-07-03:
The iterative IK solvers look for a solution within the specified constraints and starting with the current position as initial guess. If the first attempt fails, they both retry from random positions as long as the iteration and time limit is not exceeded. NloptInverseKinematics can further narrow the search space by setting upper and lower bounds, e.g., for specific postures.

Commit 0b1c027863f817c585e41bad0bdbff9d83161bfd adds the option of specifying the number of random restarts via `IterativeInverseKinematics::setRandomRestarts` and a value of zero will disable this feature.

For more detailed control of Cartesian motion on the velocity level, e.g., singularity and joint limit behavior, I would recommend to directly use the Jacobian inverse.