roboticslibrary/rl #50: how to view urdf model

george66s, 2021-11-25:
Hi, I have a robot model using urdf file format, with stl files. How can I view the model? Is there any example I can learn from?

Lots of thanks!

George

george66s, 2021-11-26:
George again. The demo shows that .xml model with .wrl files can be viewed. Just wondering if .urdf model with .stl files can be viewed or not. Thanks!

rickertm, 2021-11-26:
RL includes a UrdfFactory for loading kinematic models with rl::mdl since version 0.7. The current master branch also includes one for loading geometric models with rl::sg. You can use the rlCoachMdl on the master branch for viewing such a model by providing the same URDF file for both `SCENEFILE` and `MODELFILE1`:
```bash
Usage: rlCoachMdl SCENEFILE MODELFILE1 ... MODELFILEn
rlCoachMdl robot.urdf robot.urdf
```
The parsers support URDF files with basic or STL geometries and absolute or relative paths. File in xacro format first have to be converted to URDF files and paths with `package://` references have to be replaced by the actual paths, e.g.:
```diff
- <mesh filename="package://robot_description/meshes/link0.stl"/>
+ <mesh filename="../meshes/link0.stl"/>
```

george66s, 2021-11-28:
Lots of thank. Will try and get back...

george66s, 2021-11-28:
Here is the result. rlCoachMdl worked very well in the Clion IDE that complies the latest downloaded source in the demo sub-directory. It successfully showed the visual image of the robot model. Lots of thanks!

But it didn't work in the command line. When I typed $ rlCoachMdl Absolute_Path/my.urdf Absolute_Path/my.urdf, it returns error message: segment error (core dumped).

george66s, 2021-11-28:
Maybe it was because my lib version was not up to 0.7?

george66s, 2021-11-29:
I tried again. This time I went to the build sub-directory and used the rlCoachMdl built by Clion IDE in command line. It worked very well. Apparently something is wrong with my installed version of rlCoachMdl. Thanks.

closed 2021-11-29 by george66s