roboticslibrary/rl #52: Creating custom kinematic chain of a robot in runtime

booger-bubble, 2021-12-06:
Hello, is it possible to create custom kinematic chain of a robot model in runtime without using urdf or xml files (e.g. using just types of joints and frames/bodies, etc.)? If so, could you provide small example? Thanks :)

rickertm, 2021-12-09:
It is possible to create a robot model at runtime, this is how `rl::mdl::XmlFactory` and `rl::mdl::UrdfFactory` generate models from files. You have to create the frames (e.g., `rl::mdl::World`, `rl::mdl::Body`) and transforms (e.g., `rl::mdl::Fixed`, `rl::mdl::Revolute`) of the model tree and add them to a model instance (e.g., `rl::mdl::Dynamic`) via `Model::add()`. Afterwards, the model is initialized via `Model::update()`. I can add a small example for this to the demos.

chenguang3312, 2021-12-22:
I have the same iss，Where to find the example? Thanks

rickertm, 2021-12-22:
You can find a small example in `demos/rlDynamics1Demo/rlDynamics1Planar2Demo.cpp`. It matches the `examples/rlmdl/planar2.xml` robot definition.

closed 2022-02-14 by booger-bubble