roboticslibrary/rl #53: EET,PRM,RRT algorithm

chenguang3312, 2021-12-22:
Hi ,I know "rrt"and " prm" algorithm. what is the "EET"? 
Why use EET planning,it seems that the path is shorter?
I need to make 6axis robot trajectory as reasonable as possible.what planning algo should i choose

WangZP10969, 2021-12-27:
@chenguang3312 
It may be a related variant of EST, which is often mentioned together with PRM and RRT
EST is a tree-based motion planner that attempts to detect the less explored area of the space by measuring the density of the explored space, biasing exploration toward parts of the space with lowest density.

rickertm, 2021-12-28:
The `rl::plan::Eet` planner is an implementation of the Exploring/Exploiting Tree planner. It leverages information (e.g., from workspace connectivity) for exploitation and balances exploration and exploitation during planning. For more details on EET and a comparison against other algorithms, you can have a look at the following publications:

Markus Rickert, Arne Sieverling, and Oliver Brock. Balancing exploration and exploitation in sampling-based motion planning. IEEE Transactions on Robotics, 30(6):1305-1317, December 2014.
https://dx.doi.org/10.1109/TRO.2014.2340191
https://mediatum.ub.tum.de/doc/1281523/707307.pdf

Markus Rickert, Oliver Brock, and Alois Knoll. Balancing exploration and exploitation in motion planning. In Proceedings of the IEEE International Conference on Robotics and Automation, pages 2812-2817, Pasadena, CA, USA, 2008.
https://dx.doi.org/10.1109/ROBOT.2008.4543636
https://mediatum.ub.tum.de/doc/1289338/19583.pdf

chenguang3312, 2021-12-31:
@rickertm @WangZP10969 
Thank you for your answers，I have another question。
I need to plan the trajectory as short as possible. Is there a more suitable algorithm to use。

chenguang3312, 2021-12-31:
@rickertm @WangZP10969 
Thank you for your answers.

closed 2022-03-17 by chenguang3312