roboticslibrary/rl #55: Error when creating the robot

RodrigoCodeBernardo, 2022-01-27:
I am trying to design my robot so that when I plan the trajectory of the robot arm it does not collide with the mobile robot base. But I am getting the following errors:

First I created a test base, in which when I go to check for no collisions, it detects collisions, which makes sense because the robot is not at the predefined distance from the base. However it allows the calculation of the trajectory:

[image]

The problem is when I replace the base with the model of my mobile robot. It no longer allows me to calculate the trajectory. How can I get around the situation?

[image]

Thanks for help

rickertm, 2022-01-30:
The scene with the mobile robot is reporting a collision in body 1 of the start configuration in rlPlanDemo (see status bar). Either the self collisions are not setup correctly or there is a collision with the convex hull of the obstacle. Please note, that the collision engines will create convex hulls for polygon meshes.

The rlCollisionDemo program calls SimpleScene::isColliding() and reports collisions if any shape is colliding with any other shape. It does not take self collisions defined in a kinematic description into account, so the result cannot be compared against rlPlanDemo.

RodrigoCodeBernardo, 2022-02-02:
I have checked and there are no colliding bodies, I don't understand what the problem is.

It only works if you move the robot arm away from the base...

Thanks for help.
[robot_IDMIND.zip](https://github.com/roboticslibrary/rl/files/7987589/robot_IDMIND.zip)

rickertm, 2022-02-03:
The `IndexedFaceSet` in `robot_base.wrl` is marked as `convex TRUE`, so a convex hull is generated for this mesh which collides with the UR3 manipulator.

This geometry of the base essentially looks like this to the collision engine:
[image]

You can manually separate the base into meshes where the convex hulls fit better, e.g., bottom cylinder, top half sphere, mount, or you can use a tool for automated convex decomposition, e.g. [Boundingmesh](https://github.com/gaschler/bounding-mesh).

Some collision engines support concave meshes with reduced performance. You can modify the `IndexedFaceSet` to use `convex FALSE` to enable this.

closed 2022-02-03 by RodrigoCodeBernardo