roboticslibrary/rl #56: The question about creating a robot model

trustguan, 2022-04-18:
Thanks to your great job!
I learned to create a robot model from this site: [https://www.roboticslibrary.org/tutorials/create-a-robot-model/](url) , it said:"Three transforms are required to create one DH revolute joint: a fixed node for translation d, the revolute joint itself, and a fixed node for translation a and rotation α", but I found that between body 2 and body 3 , the description like that:
`		<fixed id="fixed4">
			<frame>
				<a idref="frame3"/>
				<b idref="body3"/>
			</frame>
			<rotation>
				<x>-90</x>
				<y>0</y>
				<z>-90</z>
			</rotation>
			<translation>
				<x>0</x>
				<y>-0.13</y>
				<z>0</z>
			</translation>
		</fixed>
`

My question is that why use "y" to make a transformation , as we know , dh paramters are related with "x" and "z"。
Further More , can you offer the robot model files that in this site:  [https://www.roboticslibrary.org/tutorials/create-a-robot-model/](url)

rickertm, 2022-04-20:
In this example, the fixed transformation node `fixed4` represents the 4x4 transform matrix that describes the transformation from the coordinate system of `body2` to the one of `body3`.

A DH transformation is given by four transform matrices: _TransZ_(d) &middot; _RotZ_(theta) &middot; _TransX_(a) &middot; _RotX_(alpha).
For this link, the values are: _TransZ_(0) &middot; _RotZ_(-90°) &middot; _TransX_(0.13) &middot; _RotX_(-90°),
which results in this matrix:
```
[ 0  0  1  0   ]
[-1  0  0 -0.13]
[ 0 -1  0  0   ]
[ 0  0  0  1   ]
```
which is equal to this transform definition in XML, where rotation is defined by Euler-ZYX angles:
```xml
<rotation>
	<x>-90</x>
	<y>0</y>
	<z>-90</z>
</rotation>
<translation>
	<x>0</x>
	<y>-0.13</y>
	<z>0</z>
</translation>
```
This image shows the coordinate systems of the links for the model and may also help in understanding the transform between these two frames:
[image]

You can find the full [kinematic](https://github.com/roboticslibrary/rl-examples/blob/master/rlmdl/comau-racer-7-14.xml) and [geometric](https://github.com/roboticslibrary/rl-examples/tree/master/rlsg/comau-racer-7-14) models for the Comau Racer 7-1.4 used in the tutorial in our [rl-examples](https://github.com/roboticslibrary/rl-examples) repository.

Please note that you do not strictly have to follow DH conventions when defining a kinematic model. The tutorial merely shows how such a definition would translate to the rl::mdl structure. You can also choose to define other rotation axes than _+Z_ via the `<axis>` node in a joint definition. This way, you can omit the rotation parts of a transform.

trustguan, 2022-04-22:
> In this example, the fixed transformation node `fixed4` represents the 4x4 transform matrix that describes the transformation from the coordinate system of `body2` to the one of `body3`.
> 
> A DH transformation is given by four transform matrices: _TransZ_(d) · _RotZ_(theta) · _TransX_(a) · _RotX_(alpha). For this link, the values are: _TransZ_(0) · _RotZ_(-90°) · _TransX_(0.13) · _RotX_(-90°), which results in this matrix:
> 
> ```
> [ 0  0  1  0   ]
> [-1  0  0 -0.13]
> [ 0 -1  0  0   ]
> [ 0  0  0  1   ]
> ```
> 
> which is equal to this transform definition in XML, where rotation is defined by Euler-ZYX angles:
> 
> ```
> <rotation>
> 	<x>-90</x>
> 	<y>0</y>
> 	<z>-90</z>
> </rotation>
> <translation>
> 	<x>0</x>
> 	<y>-0.13</y>
> 	<z>0</z>
> </translation>
> ```
> 
> This image shows the coordinate systems of the links for the model and may also help in understanding the transform between these two frames: [image]
> 
> You can find the full [kinematic](https://github.com/roboticslibrary/rl-examples/blob/master/rlmdl/comau-racer-7-14.xml) and [geometric](https://github.com/roboticslibrary/rl-examples/tree/master/rlsg/comau-racer-7-14) models for the Comau Racer 7-1.4 used in the tutorial in our [rl-examples](https://github.com/roboticslibrary/rl-examples) repository.
> 
> Please note that you do not strictly have to follow DH conventions when defining a kinematic model. The tutorial merely shows how such a definition would translate to the rl::mdl structure. You can also choose to define other rotation axes than _+Z_ via the `<axis>` node in a joint definition. This way, you can omit the rotation parts of a transform.

thanks a lot ！

closed 2022-04-22 by trustguan