roboticslibrary/rl #57: Fail due to deprecated libxml2 feature

OgreTransporter, 2022-04-29:
```
D:\RoboticsLibrarybuild\src\rl/xml/Object.h(56,12): error C2589: "constant": Invalid token on the right side of "::" [D:\RoboticsLibrary\build\src\rl\kin\kin.vcxproj]
D:\RoboticsLibrarybuildsrc\rl\kin.vcxproj]: error C2059: syntax error: "==" [D:\RoboticsLibrarybuildsrc\rl\kin.vcxproj]
D:\RoboticsLibrarybuildsrc\rl\kin.vcxproj]: error C2143: syntax error: ";" is missing before "{" [D:\RoboticsLibrarybuildsrc\rl\kin.vcxproj]
D:\RoboticsLibrary\src\src\rl/xml/Object.h(58,9): error C2039: "xmlXPtrFreeLocationSet" is not a member of "`global namespace'". [D:\RoboticsLibrary\buildsrc\rl\kin\kin.vcxproj]
D:\RoboticsLibrarybuildsrc\src\rl/xml/Object.h(58,46): error C2039: "xmlLocationSetPtr" is not a member of "`global namespace'". [D:\RoboticsLibrary\buildsrc\rl\kin\kin.vcxproj]
D:\RoboticsLibrarybuildsrc\rl\kin\kin.vcxproj]: error C2061: syntax error: identifier "xmlLocationSetPtr" [D:\RoboticsLibrarybuildsrc\rl\kin\kin.vcxproj].
D:\RoboticsLibrarybuildsrc\rl\kin.vcxproj]: error C2181: invalid "else" without associated "if" [D:\RoboticsLibrarybuildsrc\rl\kin.vcxproj]
D:\RoboticsLibrary\src\src\rl/xml/Object.h(107,12): error C2589: "constant": invalid token on right side of "::" [D:\RoboticsLibrary\build\src\rl\kin\kin.vcxproj]
D:\RoboticsLibrarybuildsrc\rl\kin.vcxproj]: error C2062: "unknown-type" type unexpected [D:\RoboticsLibrarybuildsrc\rl\kin.vcxproj].
D:\RoboticsLibrary\src\src\rl/xml/Object.h(108,12): error C2589: "constant": invalid token on right side of "::" [D:\RoboticsLibrary\build\src\rl\kin\kin.vcxproj]
D:\RoboticsLibrary\src\src\rl/xml/Object.h(109,12): error C2589: "constant": invalid token on right side of "::" [D:\RoboticsLibrary\src\rl/xml/Object.h]
```

The problem in line 56
https://github.com/roboticslibrary/rl/blob/102896d1e39459609731b7b250a1fbf40dafed7b/src/rl/xml/Object.h#L56

can be quickly understood by looking at the definition of `XPATH_LOCATIONSET` in libxml2: `#define XPATH_LOCATIONSET 7`
https://gitlab.gnome.org/GNOME/libxml2/-/blob/670701075b969962db1b869507318e81e96910d1/include/libxml/xpath.h#L120

This means `::XPATH_LOCATIONSET` becomes `::7`, which of course cannot be compiled.

See deprecation explaination: https://gitlab.gnome.org/GNOME/libxml2/-/commit/670701075b969962db1b869507318e81e96910d1

TODO: Remove/Replace XPointer locations support

rickertm, 2022-05-10:
Thank you for the heads up. I'll add a check for `LIBXML_XPTR_LOCS_ENABLED` combined with a check for `LIBXML_VERSION` once a new libxml2 version with this feature is released. Versions up to 2.9.14 do not include this definition yet.

rickertm, 2022-08-20:
Should be fixed by 471b85768127ca68d75271a41dbad1fc19fe9245.