roboticslibrary/rl #63: Gantry / CNC definition

francoislauzon, 2022-12-13:
Hello,
I am new to RL and was wondering if it is possible to define a 3 axis gantry with the library, or an elevated turn table.

Thanks,
Francois

rickertm, 2022-12-19:
The classes in rl::mdl can be used to model tree-like structures with revolute, prismatic, cylindrical, helical, spherical, and 6-DOF joint types. We had a use case with a 3-axis gantry with a 3-DOF rotational end-effector in a previous project. A 3-axis gantry can be modeled with 3 prismatic joints. How does the structure of your elevated turn table look like?

francoislauzon, 2022-12-19:
Thanks for the reply.
Can you share a simple xml of the 3-axis with 3 DOF that you have in your use case?
The turn table structure is straight forward, a simple up/down axis that move a turntable up/down.  
Thanks,
Francois

rickertm, 2022-12-19:
I am not able to share the files from the project, but I've create a basic set of geometric and kinematic models that should help you get started:
[gantry.zip](https://github.com/roboticslibrary/rl/files/10261313/gantry.zip)

If understood you correctly, then the turn table requires a prismatic joint for the up/down axis, and a revolute joint for the rotation axis of the table. If the table has additional rotations, then you have to add one or two additional revolute joints or you can directly use a spherical joint for all three.

francoislauzon, 2022-12-19:
Thank you Markus,
I load the example you gave me in rlCoachMdl and this is exactly the kind of help I was looking for.
I will analyze this, but this is an excellent starting point.

Thanks,
Francois