roboticslibrary/rl #65: How can I check if the trajectory remains valid while the robot moves?

RodrigoCodeBernardo, 2022-12-14:
How can I check if my generated trajectory is valid along the robot motion? 

The goal is to change the robot's surroundings, and if an obstacle appears on the initially calculated path, it will send an alert.

Thanks

RodrigoCodeBernardo, 2022-12-15:
There is a way to check the path through one of the functions: RecursiveVerifier or Verifier.

As inputs would have a different scenario than the one in which the plan was constructed, the robot's kinematics and the path initially obtained.

RodrigoCodeBernardo, 2022-12-15:
Hi

I found this solution, which works for now. What I do is import a new scenario and check for each point of the trajectory if it still remains feasible. There may be a better solution I am open to suggestions.

`		rl::plan::SimpleModel simpleModel;
		rl::math::Vector pose(6);	
		bool isColliding;

		simpleModel.mdl = kinematic.get(); // for rl::mdl::Kinematic*
		simpleModel.scene = &scene2;
		simpleModel.model = scene2.getModel(0); // id of kinematic model in scene
		
		optimizer.process(path_optimised);
		
		for (rl::plan::VectorList::iterator e = path_optimised.begin(); e != path_optimised.end(); ++e)
		{	
			pose = e->transpose();
			isColliding = simpleModel.isColliding(pose);
			std::cout << "isColliding() " << (isColliding ? "true" : "false") << std::endl;
		}`

closed 2022-12-15 by RodrigoCodeBernardo

rickertm, 2022-12-19:
The verifiers internally call `rl::plan::SimpleModel::isColliding`, so your approach should work for that use case. `rl::plan::SimpleModel` combines the kinematic & geometric models and filters tests of `rl::sg::SimpleScene::areColliding` between bodies based on the robot's collision definitions.