roboticslibrary/rl #67: How to add enveronment data if i using URDF file to plan path

nicu233, 2023-03-21:
Hello, thank you for your great work.
I am working for collision free robotic path plan using the URDF file.
The rlPlanDemo in the master branch is can load URDF as robotic kinematic and robotic geometry data fine.
But the URDF file does not have data about the environment. 

My question is how to add environment data to do a path plan if I use a URDF file. 
Do I need to use URDF and VRML files for robotic and enveronment separately?

rickertm, 2023-03-25:
In general, you can add additional geometries such as obstacles next to a robot inside a URDF file and combine/position them via a fixed joint relative to a root node. However, RL requires information on how these geometries relate to a robot or obstacle model. To my knowledge, the URDF file format does not yet define a common XML element for combining multiple robot definitions inside a single file.

The `rl::sg::XmlFactory` supports multiple models inside a scene file and you can use separate files for kinematics and geometry. The `rl::sg::UrdfFactory` can also load multiple robot/model definitions, there's even a loop for this:
https://github.com/roboticslibrary/rl/blob/2da1f81077247676e5bfd638db206a3ab7cb84a4/src/rl/sg/UrdfFactory.cpp#L99-L103

Although this is no standardized URDF definition, I could modify the following line to look for any `<robot>` element inside other XML elements and enable Xinclude:
https://github.com/roboticslibrary/rl/blob/2da1f81077247676e5bfd638db206a3ab7cb84a4/src/rl/sg/UrdfFactory.cpp#L89

This would allow you to define multiple `<robot>` definitions under a common XML root, e.g., `<robots>` or any other name, either inside the same file or by combining them via XInclude:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<robots xmlns:xi="http://www.w3.org/2003/XInclude">
	<xi:include href="robot.urdf"/>
	<xi:include href="obstacles.urdf"/>
</robots>
```

nicu233, 2023-03-27:
Thank you for your response. 
In the past few days, I used the rlPlanDemo to learn the robotic path plan of the RL.
For load obstacles, I changed the load function in rlPlanDemo/MainWindow.cpp to load robotic and obstacles geometry files separately.
For easy to use, I load robotic data using URDF (many robot's URDF file easy to get ), and load the environment's data using XML file which like you provided _universal-robots-ur5_tables.xml_, I just delete <robot> element, keep only <enveronment>.
I add scene load codes after your original scene load code:
`	if ("urdf" == modelSceneFilename.substr(modelSceneFilename.length() - 4, 4))
	{
		rl::sg::UrdfFactory sceneFactory;
		sceneFactory.load(modelSceneFilename, this->scene.get());
		this->sceneModel = this->scene->getModel(
			path.eval("number((/rl/plan|/rlplan)//model/model)").getValue<std::size_t>()
		);
	}
	else
	{
		rl::sg::XmlFactory sceneFactory;
		sceneFactory.load(modelSceneFilename, this->scene.get());
		this->sceneModel = this->scene->getModel(
			path.eval("number((/rl/plan|/rlplan)//model/model)").getValue<std::size_t>()
		);
	}

// my added codes, there load environment data
    rl::sg::XmlFactory sceneFactory;
    sceneFactory.load("mypath\\universal-robots-ur5_tables_test.xml", this->scene.get());
    this->sceneModel = this->scene->getModel(
        path.eval("number((/rl/plan|/rlplan)//model/model)").getValue<std::size_t>()
    );
`
It looks work for me.
[image]

closed 2023-03-27 by nicu233