roboticslibrary/rl #69: Minimal code to retrieve the joints transforms/states of a model

mohdjawadi, 2023-04-14:
Hi! Please I am using a custom model visualization/simulation framework based on C# WPF viewport3D . I would like to kindly ask how i can retrieve the state of each joint transformation so that I can reproduce the joint states in the front end. I am little bit confused on the methods getJoint(), getFrames(), and others. Does not know which one will actually return the current state of a joint. I know getOperationalPosition() only return return the state of the tool. I would like to know how to retrieve the states of the remaining joints.

Thanks

mohdjawadi, 2023-04-24:
Thank you for responding.  I have retrieve the transformation matrix using the  getBodyFrame() function but could not determine how to use this for the visualization. One major problem is determining joint origins that the a link will rotate about. Also, based on the rl::mdl::UrdfFactory codes, the joint origin has been extracted from the urdf file but added as a separate frame. So please could you assist me with the following:
(1) How to interpret the retrieved transformation matrix, like for example, to translate to Windows Presentation Foundation  [Matrix3D class](https://learn.microsoft.com/en-us/dotnet/api/system.windows.media.media3d.matrix3d?view=windowsdesktop-7.0)
(2) How can I retrieve the joint origin translational and rotational offset so I can properly visualize the motion by performing the transformation based on this offset.
Thanks

rickertm, 2023-04-24:
The transformation matrices are a typedef of [`Eigen::Transform`](https://eigen.tuxfamily.org/dox/classEigen_1_1Transform.html). You can use `Eigen::Transform::matrix()` to get a 4x4 matrix that you can convert to the Matrix3D class.
The class `rl::mdl::Joint` is a subclass of `rl::mdl::Transform` and performs a transformation of `Frame* in` to `Frame* out`, both are public members. You can access each joint via `Model::getJoint(const ::std::size_t& i)`.