roboticslibrary/rl #73: Urdf file

lgzid, 2023-10-03:
I was using the RL library to read the urdf file, but I found that m_ rlModel ->getBodies() always returns 0，here is my code：
```
    rl::mdl::UrdfFactory factory;
    std::string path1 = "/home/lzd/QT_project/catkin_my/src/package_name/urdf_Model/KUKA_KR210_R2700/DH.urdf";
    std::shared_ptr<rl::mdl::Model> m_rlModel(factory.create(path1));

    if (!m_rlModel) {
        std::cerr << "Failed to load URDF model." << std::endl;
    } else {
        std::cout << "Successfully loaded URDF model." << std::endl;
        std::cout << "Number of bodies in the model: " << m_rlModel->getBodies() << std::endl;
    }

    std::cout<<path1<<std::endl;
    std::cout<<"m_rlModel->getDof()"<<m_rlModel->getDof()<<std::endl;
    std::cout<<"m_rlModel->getBodies()"<<m_rlModel->getBodies()<<std::endl;   // 0
```

output：
KUKA_KR210_R2700
```
link: 0
	name: dummyLink
link: 1
	name: baseLink
link: 2
	name: Link1
link: 3
	name: Link2
link: 4
	name: Link3
link: 5
	name: Link4
link: 6
	name: Link5
link: 7
	name: Link6
link: 8
	name: Tool
joint: 0
	parent: dummyLink
	child: baseLink
	origin.translation: 0 0 0
	origin.rotation:  0 -0  0
	type: fixed
	name: dummyJoint
joint: 1
	parent: baseLink
	child: Link1
	origin.translation: 0 0 0
	origin.rotation:  0 -0  0
	type: revolute
	max: 180
	min: -180
	speed: 0
	axis: 0 0 1
	name: joint1
joint: 2
	parent: Link1
	child: Link2
	origin.translation:  0.35     0 0.675
	origin.rotation: -90.0002       -0        0
	type: revolute
	max: 155
	min: -94.9998
	speed: 0
	axis: 0 0 1
	name: joint2
joint: 3
	parent: Link2
	child: Link3
	origin.translation:     0 -1.15     0
	origin.rotation:       0      -0 90.0002
	type: revolute
	max: 75.0002
	min: -180
	speed: 0
	axis: 0 0 1
	name: joint3
joint: 4
	parent: Link3
	child: Link4
	origin.translation: 0.041     0     0
	origin.rotation: 90.0002      -0       0
	type: revolute
	max: 400
	min: -400
	speed: 0
	axis: 0 0 1
	name: joint4
joint: 5
	parent: Link4
	child: Link5
	origin.translation:   0   0 1.2
	origin.rotation: -90.0002       -0        0
	type: revolute
	max: 120
	min: -120
	speed: 0
	axis: 0 0 1
	name: joint5
joint: 6
	parent: Link5
	child: Link6
	origin.translation:      0 -0.215      0
	origin.rotation: 90.0002      -0       0
	type: revolute
	max: 400
	min: -400
	speed: 0
	axis: 0 0 1
	name: joint6
joint: 7
	parent: Link6
	child: Tool
	origin.translation: 0 0 0
	origin.rotation:  0 -0  0
	type: fixed
	name: joint7
root: dummyLink
Successfully loaded URDF model.
Number of bodies in the model: 0
/home/lzd/QT_project/catkin_my/src/package_name/urdf_Model/KUKA_KR210_R2700/DH.urdf
m_rlModel->getDof()6
m_rlModel->getBodies()0
```

rickertm, 2023-10-03:
You're probably using an older version of the library that only adds links with `<inertial></inertial>` entries as bodies. This behavior was modified in f6a0ec56b1003e7ba6a48c7af903d0012cab8d2a to parse all links as bodies. The current version on the master branch includes this patch.

rickertm, 2023-10-03:
The [`ppa:roblib/ppa`](https://launchpad.net/~roblib/+archive/ubuntu/ppa) PPA on LaunchPad is still using the last official release 0.7. You can use the [`ppa:roblib/daily`](https://launchpad.net/~roblib/+archive/ubuntu/daily) PPA to get the current version on the master branch.

lgzid, 2023-10-03:
@rickertm Do I need to remove the original rl 0.7 and then reinstall it by executing the following command:
```
sudo apt-add-repository ppa:roblib/daily
sudo apt-get update
sudo apt-get install librl librl-demos librl-examples
sudo apt-get install librl-dev
```

lgzid, 2023-10-03:
@rickertm Thank you very much for your patient guidance. I have successfully read my URDF file！
Thanks！！

closed 2023-10-04 by lgzid

lgzid, 2023-11-06:
@rickertm Hello! 
I found that after following the above commands to install, it shows 'rl/sg/fcl' cannot be found.

rickertm, 2023-11-17:
The Ubuntu packages built on Launchpad currently do not include FCL as a dependency (required by rl::sg::fcl), as these packages are not available on earlier Ubuntu versions. Alternatives are Bullet, ODE, PQP, and SOLID3. You can test for support by checking the preprocessor definition `RL_SG_FCL` (e.g., in rlCollisionDemo):
```cpp
#ifdef RL_SG_FCL
...
#endif // RL_SG_FCL
```