roboticslibrary/rl #74: Reading the URDF file of UR5 and calculating the initial positions resulted in an error

lgzid, 2023-10-09:
Hello! 
Recently, I've been learning Robotics Library and trying to use this library to read URDF files and correctly display a robotic arm using VTK. However, I've encountered an issue while reading the URDF file for [UR5](https://github.com/zfshenGit/urdf_ws/blob/master/urdf/ur5.urdf). It seems that the calculated positions for various components are not quite right.

Here is my code:
```
// std::shared_ptr<rl::mdl::Model> m_rlModel;
rl::mdl::UrdfFactory factory;
m_rlModel= factory.create(path1);

m_rlJointAngle = rl::math::constants::rad2deg * m_rlModel->getPosition();
    if(rl::mdl::Kinematic* kinematic = dynamic_cast<rl::mdl::Kinematic*>(m_rlModel.get())){
        kinematic->setPosition(m_rlJointAngle * rl::math::constants::deg2rad);
        kinematic->forwardPosition();
    }

// Get the x, y, z for each part.
std::vector<double> initPos;
    initPos.reserve((GetDof() + 1) * 3);
    for(size_t i = 0; i < m_rlModel->getBodies()-1; ++i){
        for(int j = 0; j < 3; ++j){
            initPos.push_back(m_rlModel->getBodyFrame(i).translation()[j]);
        }
    }

// The URDF units are in meters, while VTK uses millimeters. Here, we are performing a unit conversion.
QStringList robotData;
    for(double positionXYZ : initPos){
        robotData << QString::number(positionXYZ * 1000);
    }
QVector<QVector<double>> robotPos;
    for(int i = 0; i< m_rlModel->getBodies()-1; ++i){
        QVector<double> temp;
        for(int j=0; j<3; ++j){
            temp.push_back(robotData[i*3+j].toDouble());
        }
        robotPos.push_back(temp);
    }

// set position
for(int i = 0; i< 7; ++i){
......
         stlActor->SetPosition(robotPos[i][0], robotPos[i][1], robotPos[i][2]);

        std::cout << "(" << robotPos[i][0] << "," << robotPos[i][1] << "," <<  robotPos[i][2] << ")" << std::endl;
......
}
```

Here is my output:
```
m_rlModel->getDof()6
m_rlModel->getBodies()15
The positions of each part：
(0,0,0)  (0,0,600)  (0,0,1400) (0,0,1650) (-250,-0.000918301,1650) (-250,-0.000918301,1650) (-313.045,-0.00106059,1586.96)
```

It's evident that the positions of the first part and the second part are incorrect.
I'm not sure where the problem is. I hope to get your assistance.

Here is the visualization result：
[image]

rickertm, 2023-10-09:
The URDF model in your link is not a pure UR5 model as it has extra links and joints before the UR5 components. Here's how it looks like (using collision geometry) in rlCoachMdl:

[image]

lgzid, 2023-10-09:
Thank you for your reply! I will go study rlCoachMdl.

lgzid, 2023-10-09:
Thank you for your reply! I will go study rlCoachMdl and use pure UR5 urdf model.

closed 2023-10-14 by lgzid