roboticslibrary/rl #75: Customizing Initial Position and Implementing Inverse Kinematics for the Robotic Arm

lgzid, 2023-11-23:
Hello, I am using the robotics library for motion planning of a mobile robotic arm. 

My robotic arm is imported through a URDF file, but the initial position of the entire arm is determined by the URDF upon import. Subsequent inverse kinematics calculations are based on this initial position. 

Currently, I need to move the robotic arm to **a specified position** after importing, for example, from the initial position (0,0,0) to (10,0,0), and then perform inverse kinematics. Is there a method in this library to achieve this?

rickertm, 2023-12-15:
As you mention a URDF file, I assume you are using either `rl::mdl::JacobianInverseKinematics` or `rl::mdl::NloptInverseKinematics` for solving the inverse kinematics of your robot arm? These implementations work on a `rl::mdl::Kinematic` model and their `solve()` function uses the current joint position of the model as starting point for their iterative approach. This position can be set before solving via `rl::mdl::Model::setPosition`. Please note that these iterative solvers retry with a random joint position in case the iterations do not produce a valid solution within their limits. This can be disabled via `rl::mdl::IterativeInverseKinematics::setRandomRestarts`.

lgzid, 2023-12-15:
Thank you for your response. I have implemented the above functionality using the` ::rl::math::Transform& world(); `function.
Thanks!

closed 2023-12-21 by lgzid