roboticslibrary/rl #78: #include <rl/sg/XmlFactory.h>error,not exist XmlFactory.h in sg file

philchenup, 2023-12-14:
when i try rlSceneCollisionTest.cpp demo file in visual studio 2019,  no other error happen, but find #include <rl/sg/XmlFactory.h>error in  mode debug x64 enviorment.finally, i found XmlFactory.h in rl\mdl file, is that normal.

rickertm, 2023-12-15:
From your description it looks like you are mixing the 0.7 version of the library with the current master branch? The `rl::sg::XmlFactory` class is not present in version 0.7 and loading a scene file was performed via `rl::sg::Scene::load`:
https://github.com/roboticslibrary/rl/blob/e4100314106124d2c629e3e280452c02ebd08397/tests/rlCollisionTest/rlSceneCollisionTest.cpp#L83-L87

On the current master branch, the `rl::sg::XmlFactory` class  was introduced and the (now deprecated) `rl::sg::Scene::load` function creates such an instance. Here are the matching lines for `rlSceneCollisionTest.cpp`:
https://github.com/roboticslibrary/rl/blob/67dddab6eafaea4d0595ccf1b43aca26e500a117/tests/rlCollisionTest/rlSceneCollisionTest.cpp#L85-L91

philchenup, 2023-12-15:
thanks for you clear up my confusion.

closed 2023-12-15 by philchenup