roboticslibrary/rl #83: Could this be packaged?

Kreijstal, 2024-02-12:
Like is there any interest in this being packaged in debian for linux or msys2

rickertm, 2024-02-13:
I think packaging RL would be quite interesting for a lot of users. We have spent some effort in adding missing dependencies in MSYS2 or vcpkg and there is a Launchpad repository ([stable](https://launchpad.net/~roblib/+archive/ubuntu/ppa) and [daily](https://launchpad.net/~roblib/+archive/ubuntu/daily)) with packages for Ubuntu (including [build instructions for Debian](https://github.com/roboticslibrary/rl-debian)), but we have not worked on packaging RL itself. Any help on this would be greatly appreciated.

Kreijstal, 2024-02-13:
I think most distros package only if there are number releases, for example on github tags and so on, for example debian gets a release and only upgrades the minimum version but no major numbers in the interest of stability and compatibiliy, furthermore I know msys would package this if a tagged version would run, compile and work :)

rickertm, 2024-02-17:
Thank you for the feedback, a new tagged version is indeed long overdue. We'll try to focus on that in the next few weeks.