roboticslibrary/rl #84: How to compute a cartesian path?

akjay, 2024-02-17:
Is there a function to compute a cartesian path like moveit --computeCartesianPath().

rickertm, 2024-02-17:
You can calculate a joint configuration for a goal pose via one of the IK solvers in rl::mdl, but the currently included path planning algorithms do not restrict the robot's motion to a Cartesian path if that is what you are looking for. In particular the RRT planners can be extended to restrict the robot's motion given specified constraints, but this will limit the planner's overall performance.

You can take a look at the included EET planner implementation, that uses workspace information and the robot's Jacobian matrix to step toward Cartesian positions as part of its exploitation phase.

akjay, 2024-02-19:
Thank you.I will try to make a cartesian path planner by inherit from rl::plan::planner