roboticslibrary/rl #85: How to change the visualization tool？

akjay, 2024-02-22:
RL give us a visualization tool by coin3d OIV,in some case,I want change the visualization tool to other type  so it can compatible other data forms such as point cloud,stl,so what I should do to achieve this,is there a guide tell users how to do it.

rickertm, 2024-03-09:
If you want to visualize your scenario with a different framework, you can get the individual coordinate frames for the bodies of a robot and the obstacles from the kinematic model and use them to update your geometries.

The rl::sg part of RL serves as an abstraction layer for a scene graph representation. It uses the structure from Open Inventor for representing geometries and adapts these to other collision detection frameworks. You can also add an abstraction for your visualization framework here. For STL files you can have a look at [SoSTLFileKit](https://www.coin3d.org/Coin/html/classSoSTLFileKit.html) from Coin3D, this is used in rl::sg::UrdfFactory.

akjay, 2024-03-20:
I decide to change visualization tool to OIV,it is easy to show point cloud with OIV,will try other tool later.

closed 2024-03-20 by akjay