roboticslibrary/rl #86: Segmentation fault occurred using rlPlanDemo

shuzhenglin, 2024-03-09:
Hi,I am researching motion planning for robotic arms and trying to figure out how to use urdf working with RL.I have installed version 0.7.0 before,but i find rlPlanDemo.c​​pp in the current version on the master branch have already support urdf file,and run this demo using the cmd as follow:
./rlPlanDemo --enable-quit /data/motion_planning/kuka_kr10_r1100-7/urdf/kuka_kr10_r1100-7.urdf
It comes out:
Segmentation fault (core dumped)
But when i use the demo rlCoachMdl ,it works
I believe that my urdf file and the calling method for stl files is correct
Is there a problem with my function usage​​
[image]
I am looking forward to your reply,and thanks for help!

rickertm, 2024-03-09:
rlPlanDemo requires a scenario description in XML format as described in the [Specify a Path Planning Scenario](https://www.roboticslibrary.org/tutorials/specify-a-path-planning-scenario/) tutorial on our website, see also the corresponding [XML schema](https://github.com/roboticslibrary/rl/blob/master/examples/rlplan/rlplan.xsd). For the filenames of the kinematic and geometric models in that description you can then specify a URDF file.

shuzhenglin, 2024-03-11:
Thanks for the fast reply and your time. I will try this and go through the rlPlanDemo again.

shuzhenglin, 2024-03-13:
Hi,I have some more questions to ask you,when i go through the rlPlanDemo,i can't find some code about QAction for MousePressEvent,It is some dropdown boxes are shown in the figure,could you tell me which file these codes are located in.
[image]
Thanks again :smile: ! @rickertm

shuzhenglin, 2024-03-16:
Hello,my friend,i am trying to use RL to operate a real KUKA robot,I have code a program using xml to communicate before,by this way,i need to write relevant program in the control cabinet as well.when i go through the rlAxisControllerDemo,I am little confused about this piece of code.Can you help me answer my confusion.Here is the partial code for the Coach.cpp
			
		      ``this->out << 6 << " " << this->i << ::std::endl;
			
			this->socket.send(this->out.str().c_str(), this->out.str().length());
			
			this->out.clear();
			this->out.str("");
			
			this->in.fill(0);
			this->socket.recv(this->in.data(), this->in.size());
			
			::std::this_thread::sleep_until(start + this->getUpdateRate());``
I don't know What is variable 'i' represent,and `` this->out << 6 << " " << this->i << ::std::endl; `` 
What does the output stream sent to the robot by this code represent，whether need the relevant code of the control cabinet to cooperate with the operation.
Hope for your reply.

rickertm, 2024-03-31:
The dropdown boxes are part of the SoQt framework's default viewers, e.g., [SoQtExaminerViewer](https://www.coin3d.org/SoQt/html/classSoQtExaminerViewer.html).

The `rl::hal::Coach` class is used as an interface to the rlCoachKin and rlCoachMdl visualization tools for sending and receiving joint configurations. It uses a simple text-based format for exchanging joint configurations and accessing individual transforms of bodies and shapes. You can have a look at `Socket::readClient()` in these two demos for more information.

For controlling a real Kuka industrial robot, you require an implementation of the real-time interface for that specific protocol. One option for the Kuka is using the Robot Sensor Interface (RSI) that requires an extra technology package from the manufacturer. The Kuka iiwa robot uses the optional Fast Robot Interface (FRI).