roboticslibrary/rl #87: Collision check with URDF file

akjay, 2024-03-14:
I am trying to use RL to do some path planning work,since I have used ROS MoveIt! in ubuntu16.04 before,so I choose URDF file as my robot description file,so I can use these file both in RL and ROS ,no need to keep two different version.I compiled latest verison of  RL library to get the URDF parse attribute.
But when I trying to solve the planning problem,it tell me there is a collision in start or goal pose,but I am sure that no collision in the given pose,so I changed the collision library from SOLID to BULLET,it worked,that no collision found and path plan successed.But after I try some different pose I found that BULLET consider every pose is collision free even pose like the picture shown,but it certainly wrong.
[image]

So it is strange that BULLET tells every pose is collision free but other collision library like PQP/ODE/SOLID tells every pose is in collsion,but this URDF file is work well in Moveit!So I don't why it can't be used in RL.

By the way,the attach file show the demo I used with rlPlanDemo,and if I changed the planner from rrtCon to rrt,no path will be found even with BULLET,it seemed that the planner is searching the path very very long time and never reach the goal.

Hope your reply.
[IRB1520.zip](https://github.com/roboticslibrary/rl/files/14599659/IRB1520.zip)

rickertm, 2024-03-31:
The reason for the collisions in all configurations is due to the missing information about self collision between the links. By default, all links are assumed to be colliding with each other. The native rl::mdl format includes the `<ignore idref="">` tag for disabling collisions between links. The SDF format also includes a [self_collide](http://sdformat.org/spec?ver=1.11&elem=link#link_self_collide) tag and specifies that links connected by a joint will never collide, which may not always be correct. You can manually define self collision between bodies via `rl::mdl::Body::setSelfCollision(Body* body, const bool& collision)`.

The issue with Bullet is due to defaulting to non-convex geometries (and using `btTriangleIndexVertexArray` over `btConvexHullShape`) as some meshes I tested had issues during the generation of convex hulls:
https://github.com/roboticslibrary/rl/blob/0b3797215345a1d37903634095361233d190b2e6/src/rl/sg/UrdfFactory.cpp#L415

akjay, 2024-12-05:
> The reason for the collisions in all configurations is due to the missing information about self collision between the links. By default, all links are assumed to be colliding with each other. The [...]

Thank you,after define self collosion between bodies as you said,the planner worked with my urdf file for the engine solid.

closed 2024-12-05 by akjay