roboticslibrary/rl #88: Inverse solution ---Tool coordinates

jinzhouchu, 2024-03-30:
Hello！
        When using the RL library for inverse kinematics, I found that the rotation of joint "c" is not aligned with the direction of the tool axis. Do I need to modify any parameters? I intend to use the tool coordinate to control the robotic arm.
        The demo I referred to is rlCoachMdl.
[image]

jinzhouchu, 2024-03-30:
When I adjust the C value, the robotic arm rotates around the blue line.

rickertm, 2024-03-31:
The values in the operational widget refer to the operational position and orientation of the robot's end effector in world coordinates. In order to specify changes in the tool frame, you can calculate the robot's Jacobian matrix in tool coordinates, see [rl::mdl::Kinematic::calculateJacobian()](http://doc.roboticslibrary.org/0.7.0/df/d55/classrl_1_1mdl_1_1_kinematic.html#a3479ee40f2603eb5fe22ca1c420f2d39).

jinzhouchu, 2024-04-01:
Thank you for your answer, but I'm a beginner in RL. Could you please guide me on how to modify this code to achieve the desired effect? I appreciate your assistance.

`
bool RLConvertAPI::SetIndexedInverseValue(const int &index, const double &value)
{
    if (rl::mdl::Kinematic* kinematic = dynamic_cast<rl::mdl::Kinematic*>(aReaderAPI->JointModel.get()))
    {

        rl::math::Transform transform = kinematic->getOperationalPosition(0);

        rl::math::Vector3 orientation = transform.linear().eulerAngles(2, 1, 0).reverse();

        switch (index)
        {
        case 0:
        case 1:
        case 2:
            transform.translation()(index) = value;
            break;
        case 3:
            transform.linear() = (
                        rl::math::AngleAxis(orientation.z(), rl::math::Vector3::UnitZ()) *
                        rl::math::AngleAxis(orientation.y(), rl::math::Vector3::UnitY()) *
                        rl::math::AngleAxis(value * rl::math::constants::deg2rad, rl::math::Vector3::UnitX())
                        ).toRotationMatrix();
            break;
        case 4:
            transform.linear() = (
                        rl::math::AngleAxis(orientation.z(), rl::math::Vector3::UnitZ()) *
                        rl::math::AngleAxis(value * rl::math::constants::deg2rad, rl::math::Vector3::UnitY()) *
                        rl::math::AngleAxis(orientation.x(), rl::math::Vector3::UnitX())
                        ).toRotationMatrix();
            break;
        case 5:
            transform.linear() = (
                        rl::math::AngleAxis(value * rl::math::constants::deg2rad, rl::math::Vector3::UnitZ()) *
                        rl::math::AngleAxis(orientation.y(), rl::math::Vector3::UnitY()) *
                        rl::math::AngleAxis(orientation.x(), rl::math::Vector3::UnitX())
                        ).toRotationMatrix();
            break;
        default:
            break;
        }

        rl::math::Vector q = kinematic->getPosition();

        rl::mdl::InverseKinematics* aInverse = new rl::mdl::JacobianInverseKinematics(kinematic);

        aInverse->addGoal(transform, 0);

        bool solved = aInverse->solve();

        if (solved)
        {
            rl::math::Vector currentPos = kinematic->getPosition();

            SetJointValue(currentPos);

            delete aInverse;

            return true;

        }
        else
        {
            kinematic->setPosition(q);

            kinematic->forwardPosition();

            delete aInverse;

            return false;
        }
    }
    return false;
}

`

jinzhouchu, 2024-04-02:
Looking forward to your reply